thesofproject/sof-bin #9: Cover what version we're supposed to use?

EvanCarroll, 2020-05-24:
Background, bought a new X1 Carbon Gen 7, with 10th Generation Intel. [Having some problems with sound](https://unix.stackexchange.com/questions/586914/audio-errors-direct-firmware-load-for-intel-sof-sof-cml-ri-failed-with-error-2). Running Debian Buster which uses the 5.6 Kernel. More information on the link above.

----

It's not clear to me coming here to install/update firmware which version of sof I should be using. Presumably the abis of these firmwares are linked to the kernel module that loads them. I don't see it documented what SOF is supported with specific kernel number. That information would be **great** to have in the README.

EvanCarroll, 2020-05-24:
It seems that my above suspicion was correct, when I try to load v.15 of SOF in the kernel 5.6, I get

    [   11.722865] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:16:0 Kernel ABI 3:12:0

paulstelian97, 2020-05-25:
Do note that there is a bit of leeway (we don't get MAJOR ABI changes often, and MINOR changes allow for compatibility which means you can run a newer firmware under an older kernel or viceversa just fine.

lgirdwood, 2020-05-25:
@EvanCarroll generally users should be using the version shipped with their distro, but you are right we do need some information for users who would like to update their kernels or FWs.  i.e. it should work but YMMV

EvanCarroll, 2020-05-25:
Good info. Using the version the hardware ships is ideal, if supported. My [distro (Debian) shipped firmware that didn't actually work with the hardware](https://wiki.debian.org/InstallingDebianOn/Thinkpad/X1%20Carbon%207th%20Gen). So we're all forced to do this for now.

lgirdwood, 2020-05-27:
@EvanCarroll Debian instructions look good, btw not sure if you also need UCM files too. Maybe the instructions could be updated to just pull the sof-bin git repo in the future once the go.sh is updated. That way folks could use v1.6 etc once released.

marc-hb, 2021-10-05:
@plbossart does the sof-bin URL now get printed when there is a MAJOR mismatch? https://github.com/thesofproject/linux/pull/2639

plbossart, 2021-10-05:
IIRC we throw an error only when request_firmware() fails. We haven't had any MAJOR ABI changes so far.

marc-hb, 2021-10-06:
I found that all these recommend a minimum kernel version which I think addresses the issue. I will close it if there's no objection.

https://github.com/thesofproject/sof/releases/tag/v1.7
https://github.com/thesofproject/sof/releases/tag/v1.8
https://github.com/thesofproject/sof/releases/tag/v1.9

closed 2021-10-29 by marc-hb