thesofproject/sof-bin #21: Missing git version tag

kloczek, 2020-10-15:
Looks like in the repo there is no at all any version tags.

heftig, 2020-10-31:
Yeah, just having branches makes it really hard to create reproducible packages. Please use tags.

lgirdwood, 2020-11-03:
ok, the reasoning behind the branch was that we could keep the repo small by deleting old branches, but I do agree we should have a tag corresponding to the binaries on each branch. @heftig @kloczek would this work for you ?

heftig, 2020-11-03:
> ok, the reasoning behind the branch was that we could keep the repo small by deleting old branches,

You can also delete tags if you want to reduce the repo size.

> we should have a tag corresponding to the binaries on each branch

I'm not sure what you mean by "a tag corresponding to the binaries".

---

My main issue is that your branches move: Last month you added commits adding `sof-tgl.ri` and a tplg file to the `stable-v1.6` branch, causing the contents of https://github.com/thesofproject/sof-bin/archive/stable-v1.6.tar.gz to change. So `v1.6` is not stable.

I could download https://github.com/thesofproject/sof-bin/archive/08a6b0f51e441f435e2c1d62c0e2063cd770841e.tar.gz instead, but that is less obvious and more of a maintenance burden than using `v1.6`.

I would expect a tag on every commit you expect downstream to use. Since `sof-bin` apparently can release more frequently than `sof` itself, I would have used a tag like `v1.6-1` on d211dd04a1cf0bd771c5c6727e7e929ea1f8909d and `v1.6-2` on 08a6b0f51e441f435e2c1d62c0e2063cd770841e.

---

That said, I think what GitHub actually expects you to do is to take what you've placed at https://github.com/thesofproject/sof-bin/tree/stable-v1.5.1 and attach it as a release asset to https://github.com/thesofproject/sof/releases/tag/v1.5.1.

lgirdwood, 2020-11-03:
> > ok, the reasoning behind the branch was that we could keep the repo small by deleting old branches,
> 
> You can also delete tags if you want to reduce the repo size.

But this shouldn't delete history, i.e. if I deleted v1.6 tag then the commits should still exist (given a new v1.7, v1.8  etc is tagged on top of those commits ) ?  The branch idea was also about protecting any common infrastructure on master.
> 
> > we should have a tag corresponding to the binaries on each branch
> 
> I'm not sure what you mean by "a tag corresponding to the binaries".

I means tagging the HEAD of v1.6 branch for v1.6-rc1, rc2 etc.
> 
> My main issue is that your branches move: Last month you added commits adding `sof-tgl.ri` and a tplg file to the `stable-v1.6` branch, causing the contents of https://github.com/thesofproject/sof-bin/archive/stable-v1.6.tar.gz to change. So `v1.6` is not stable.
Sorry about this, v1.6 is not released yet (we are still on rc3 ) but hopefully have resolved the last issue now and pending validation for v1.6 final. 

> 
> I could download https://github.com/thesofproject/sof-bin/archive/08a6b0f51e441f435e2c1d62c0e2063cd770841e.tar.gz instead, but that is less obvious and more of a maintenance burden than using `v1.6`.
> 
> I would expect a tag on every commit you expect downstream to use. Since `sof-bin` apparently can release more frequently than `sof` itself, I would have used a tag like `v1.6-1` on [d211dd0](https://github.com/thesofproject/sof-bin/commit/d211dd04a1cf0bd771c5c6727e7e929ea1f8909d) and `v1.6-2` on [08a6b0f](https://github.com/thesofproject/sof-bin/commit/08a6b0f51e441f435e2c1d62c0e2063cd770841e).
> 
> That said, I think what GitHub actually expects you to do is to take what you've placed at https://github.com/thesofproject/sof-bin/tree/stable-v1.5.1 and attach it as a release asset to https://github.com/thesofproject/sof/releases/tag/v1.5.1.

@heftig thanks for the inputs, lets make sure we get this right for v1.6 final (the rc lasted a lot longer than we wanted due to some bugs on some HW platforms).  

@mengdonglin @xiulipan can you own this moving forwards ?

heftig, 2020-11-03:
> But this shouldn't delete history, i.e. if I deleted v1.6 tag then the commits should still exist (given a new v1.7, v1.8 etc is tagged on top of those commits ) ? The branch idea was also about protecting any common infrastructure on master.

If `v1.7` depends on the commit tagged `v1.6` then deleting the latter won't help, of course. But that doesn't have to be the case; you can keep your current branches, they just need tags, too.

You can also push tags that aren't part of any branches. That's not often done, though, and `git clone` will not fetch such tags by default. Perhaps you would prefer that, though.

xiulipan, 2020-11-05:
@lgirdwood @heftig Will try to update with tags with v1.6 final release.
PS: Do we need to add old TAG for previous branches and releases?
PPS: what tag style would be used here? I am not sure if NXP will upload their binaries after we tag one.

dbaluta, 2020-11-05:
@xiulipan I have not followed all the conversation but for NXP we are relasing our binaries using our Yocto BSP and we create our separate branch from stable (e.g imx-stable-v1.6).

xiulipan, 2020-11-05:
@dbaluta Thanks for the clarify. Then I think we can tag our branches independently for v1.6

dbaluta, 2020-11-05:
@xiulipan NXP will use whatever releasing scheme is decided. I have now read the entire thread and using tags for reproducible builds makes sense. 

Not sure who should add those tags. Are the tags are to the code repo or to the binary repo?

xiulipan, 2020-11-19:
Tags is added for v1.6 and v1.6-rc3.

closed 2020-11-19 by xiulipan

heftig, 2021-04-05:
`master` now contains v1.7. Shouldn't this be tagged?

lgirdwood, 2021-04-06:
@heftig thanks, your right it's missing a tag. Xiulipan used to do this and has recently joined a different company. I've now tagged and updated the readme.
@keyonjie fyi - you will need do todo the tag and readme.md update on future releases.