thesofproject/sof-bin #22: Installation is a nightmare for the naive user

BrucePerens, 2020-11-02:
Hi,

I am currently running Debian stable with a bleeding-edge kernel. What does it take to get the SOF driver running?

1. Look all over the net, and mostly get answers that you must turn off the digital microphone in the old HDA driver. Wrong.
2. Look through dmesg and notice that a firmware file isn't loading. Look in /lib/firmware, it's not there.
3. Install the latest firmware from the linux-firmware repository. Nope, not in there either.
4. Find the SOF project.
5. Look in the latest releases. Only source available.
6. Find sof-bin. Download the master, because that's git's default.
7. Run go.sh . Doesn't work with the master.
8. Look for the name of the latest signed release. Can't find that.
9. Checkout the arbitrary old release mentioned in an aside in the documentation.
10. Run go.sh again. It works.

Why isn't this in the linux-firmware repository?

    Thanks

    Bruce

plbossart, 2020-11-02:
@BrucePerens using linux-firmware was the initial plan but there were disagreements on the 'debug tools (sof-logger and .ldc files), so we provided sof-bin as is and assumed distro maintainers would pick the firmware from there. that's the working model with Fedora and Ubuntu.
We've had email conversations with some Debian maintainers but no one picked up the task of doing the packaging.

That said you're right that we should do a better job advertising what the latest version is, and add more references on this in the  documentation https://thesofproject.github.io/latest/getting_started/index.html#debugging-audio-issues-on-intel-platforms

BrucePerens, 2020-11-02:
For now, it would be nice if the internal disagreement could be resolved to the point that production firmware - not the debug tools - could be placed on the linux-firmware repository. People who build their own kernels can't really depend on distribution packagers to keep up with the kernel driver, but they have learned to update from the firmware repository.

marc-hb, 2020-12-01:
@plbossart does everyone need the `sof-logger` and `.ldc` files?

plbossart, 2020-12-01:
@marc-hb that'd be the first thing we ask if there is a complicated bug report where we need a trace... So not all users need this, but power users who report bugs need to have an 'apt install' way to getting these files.

marc-hb, 2020-12-01:
Totally understand the requirement for an`apt install` level of user friendliness but it sounds like we could release the firmware and `.tplg` files only to `linux-firmware`, no? Now this would assume a bulletproof way to match them to the corresponding logger files when needed and I guess we don't really have that yet.

plbossart, 2020-12-01:
There was also the argument that topology files aren't binaries and as such not suitable for linux-firmware - they can be re-generated with alsa-lib...

marc-hb, 2020-12-02:
Now that last bit about .tplg files sucks more, do you have a pointer, approx. date and/or some search keywords to that past argument? 

`hexdump -C` shows many ASCII characters in .tplg files but for me they're binaries until someone demonstrates that `cat` can send them to some locale and terminal without wrecking the latter.

So `linux-firmware` would be only for binaries that can't be regenerated from open source? I don't see that in its README.

It would be error-prone to request all Linux distributions to match the version of the topology sources with the version of the firmware binary. It would also be a waste of effort to ask them to regenerate the (hopefully) exact same .tplg files independently...

cc: @lgirdwood

BrucePerens, 2020-12-03:
It sounds to me like you guys are overthinking this. I believe you should
put all the files that it takes to make the driver work in the
linux-firmware distribution. If someone pushes back, you can remove files
then. But I doubt anyone will. If you do that, those files will get into
distributions, and they will be used by everyone who builds a custom
kernel. Right now I believe some distributions, and a lot of people who
build custom kernels, are holding back on configuring new kernels to use
your driver; because it results in a non-working sound system.

Just make it simple for the user.

Thanks

Bruce

lgirdwood, 2020-12-03:
@BrucePerens you are right, there was some pushback on including topology binaries and logger strings (the LDC files) alongside the firmware binaries in linux-firmware. The driver needs both the FW binary AND the topology binary to work. Only the logger needs the firmware string (LDC file). 

@marc-hb topologies are binaries built from ASLA conf public sources, but you are right in that it's not easy for distro's to easily build the topologies (as may need ALSA lib/utils git versions from time to time) or easily build  the firmware (may need proprietary compiler) and notwithstanding any code signing for platforms with private keys. 

@marc-hb can we look at linux-firmware again for FW binary and topologies (not LDC files). @perexg would this work for you, IIUC you had some issues with the LDC files being present in linux-firmware but I can't recall now if you had a similar issue with the topology binaries ?

I've also disabled force push rebasing in this repo.

perexg, 2020-12-03:
> @marc-hb can we look at linux-firmware again for FW binary and topologies (not LDC files). @perexg would this work for you, IIUC you had some issues with the LDC files being present in linux-firmware but I can't recall now if you had a similar issue with the topology binaries ?

We package the SOF firmware in the separate package alsa-sof-firmware at the time using sof-bin. The linux-firmware is going to be so big anyway, but it's probably out-of-scope to resolve this issue here.

It may make sense to mirror the latest necessary files to linux-firmware, but our linux-firmware package will not use those files (keeping them in the alsa-sof-firmware). Also, think about driver / topology / DSP code compatibility issues. The linux-firmware is supposed to work with wide range of kernel versions and the SOF is (or was?) under the heavy development.

Past discussion: https://mailman.alsa-project.org/pipermail/alsa-devel/2019-May/149961.html [last paragraph]

BrucePerens, 2020-12-03:
If you absolutely can't put the required packages in linux-firmware right
now, please have the error message give the proper URL for the firmware
package when it can't load the firmware. It took me at least half an hour
to figure out where to get it, as detailed in the opening message.

plbossart, 2020-12-03:
@marc-hb the notion of topology predates SOF, and there isn't a single topology file that was added to linux-firmware. I don't have a trace but a very good memory of a discussion with the Skylake driver owners a couple of years ago - but it doesn't hurt to ping the linux-firmware maintainers for confirmation.

I also want to mention that we have a ton of topology files, see [1], compared to a single firmware binary per hardware generation.  I really don't see how we could possibly keep every version of topology file in linux-firmware, as currently done with the symbolic links.

[1] https://github.com/thesofproject/sof/blob/cc1cd8a4af1b4b6c87f9b9c1da055f96421dde28/tools/topology/CMakeLists.txt#L19

marc-hb, 2020-12-17:
Hi @BrucePerens 
- printing the URL when the firmware is missing has been queued, see above
- We will submit firmware to linux-firmware and topologies if acceptable (afraid not)
- I started work to simplify the go.sh and packaging experience.

Please let us know if I missed anything.

BrucePerens, 2020-12-17:
I'll try a new kernel. Thanks!

LarsImNetz, 2021-01-24:
root@phoebe /home/lars/develop/hardware-lenovo/audio/sof-bin-1.6.1 # SOF_VERSION=1.6.1 ./go.sh
Can't find version 1.6.1 - are you missing leading v?

Why must it be so complex?

How to build 1.6.1 release in the right way?

mmokrejs, 2021-04-09:
BTW, did you guys want to state in the README.md that one is supposed to enable in .confg at least the following?

```
CONFIG_SND_SOC_SOF_TOPLEVEL=m
SND_SOC_SOF_PCI=m
SND_SOC_SOF_ACPI=m
```

```
  Sound Open Firmware Support (SND_SOC_SOF_TOPLEVEL) [Y/n/?] y
    SOF PCI enumeration support (SND_SOC_SOF_PCI) [M/n/?] m
    SOF ACPI enumeration support (SND_SOC_SOF_ACPI) [M/n/?] m
    SOF nocodec mode support (SND_SOC_SOF_NOCODEC_SUPPORT) [N/y/?] n
    SOF strict ABI checks (SND_SOC_SOF_STRICT_ABI_CHECKS) [N/y/?] n
    SOF debugging features (SND_SOC_SOF_DEBUG) [N/y/?] n
    SOF support for Intel audio DSPs (SND_SOC_SOF_INTEL_TOPLEVEL) [N/y/?] (NEW) ?
      SOF support for Baytrail, Braswell and Cherrytrail (SND_SOC_SOF_BAYTRAIL_SUPPORT) [N/y/?] (NEW) 
      SOF support for Broadwell (SND_SOC_SOF_BROADWELL_SUPPORT) [N/y/?] (NEW) 
      SOF support for Tangier/Merrifield (SND_SOC_SOF_MERRIFIELD_SUPPORT) [N/y/?] (NEW) 
      SOF support for Apollolake (SND_SOC_SOF_APOLLOLAKE_SUPPORT) [N/y/?] (NEW) 
      SOF support for GeminiLake (SND_SOC_SOF_GEMINILAKE_SUPPORT) [N/y/?] (NEW) 
      SOF support for Cannonlake (SND_SOC_SOF_CANNONLAKE_SUPPORT) [N/y/?] (NEW) 
      SOF support for CoffeeLake (SND_SOC_SOF_COFFEELAKE_SUPPORT) [N/y/?] (NEW) 
      SOF support for Icelake (SND_SOC_SOF_ICELAKE_SUPPORT) [N/y/?] (NEW) 
      SOF support for CometLake-LP (SND_SOC_SOF_COMETLAKE_LP_SUPPORT) [N/y/?] (NEW) 
      SOF support for CometLake-H (SND_SOC_SOF_COMETLAKE_H_SUPPORT) [N/y/?] (NEW) 
      SOF support for Tigerlake (SND_SOC_SOF_TIGERLAKE_SUPPORT) [N/y/?] (NEW) 
      SOF support for ElkhartLake (SND_SOC_SOF_ELKHARTLAKE_SUPPORT) [N/y/?] (NEW) 
```

Does the above apply even if one has a USB-based card?

Somebody pointed me from https://askubuntu.com/questions/1218058/ubuntu-19-10-audio-driver-issue/1238287#1238287  to this project due to a driver missing for UGREEN USB sound dongle  https://www.amazon.de/-/en/gp/product/B087T5H3MQ based on Realtek ALC4030 chipset. So what about such card connected via USB?

```Apr  9 17:08:55 dell_e5580 kernel: usb 1-2.3.3: new full-speed USB device number 26 using xhci_hcd
Apr  9 17:08:56 dell_e5580 kernel: usb 1-2.3.3: New USB device found, idVendor=0bda, idProduct=48f0, bcdDevice= 0.03
Apr  9 17:08:56 dell_e5580 kernel: usb 1-2.3.3: New USB device strings: Mfr=3, Product=1, SerialNumber=0
Apr  9 17:08:56 dell_e5580 kernel: usb 1-2.3.3: Product: USB Audio
Apr  9 17:08:56 dell_e5580 kernel: usb 1-2.3.3: Manufacturer: Generic
Apr  9 17:08:56 dell_e5580 kernel: input: Generic USB Audio as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.3/1-2.3.3/1-2.3.3:1.0/0003:0BDA:48F0.0008/input/input25
Apr  9 17:08:56 dell_e5580 kernel: hid-generic 0003:0BDA:48F0.0008: input,hidraw4: USB HID v1.11 Device [Generic USB Audio] on usb-0000:00:14.0-2.3.3/input0
```

```
# lsusb -d 0bda:48f0 -v -v

Bus 001 Device 026: ID 0bda:48f0 Realtek Semiconductor Corp. USB Audio
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x48f0 
  bcdDevice            0.03
  iManufacturer           3 Generic
  iProduct                1 USB Audio
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      57
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x88  EP 8 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               4
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

plbossart, 2021-04-09:
@mmokrejs your question is unrelated to the firmware releases really, what people enable in their distributions is their choice entirely. We provide a list of configs that we used in our tests:
https://github.com/thesofproject/kconfig/blob/master/sof-defconfig
https://github.com/thesofproject/kconfig/blob/master/sof-dev-defconfig

and we provide guidelines here 
https://thesofproject.github.io/latest/getting_started/setup/setup_ktest_environment.html#save-your-kernel-config-as-ktest-sof-dev-defconfig

marc-hb, 2021-09-28:
@BrucePerens no news is good news?

closed 2021-10-04 by marc-hb