thesofproject/sof-bin #25: The update nightmare in the stable-v1.6 branch

perexg, 2020-11-30:
Can someone explain me the updates in the stable-v1.6 branch?

The HEAD commit in stable-v1.6 is floating. We are using this as the source of our packages. Please, do incremental changes only or do not mark your code as stable (put it to the stable branch)! Thank you.

Commit da8cb04 disappeared like commit 47b436a .

@plbossart , @xiulipan - comments?

perexg, 2020-11-30:
Can I use the commit cbdec6963b2c2d58b0080955d3c11b96ff4c92f0 as the final binary v1.6 release? Or not?

plbossart, 2020-11-30:
@lgirdwood @mwasko this one is for you to address.

xiulipan, 2020-12-01:
@perexg I think we also have rc binary in this branch. To avoid big git size, we always rebase the branch.
@lgirdwood Should we use tag for the release or branch?

perexg, 2020-12-01:
> @perexg I think we also have rc binary in this branch. To avoid big git size, we always rebase the branch.

The development should be moved to some other place or just do incremental changes. I have packages which refers to the non-existent commits da8cb04 and 47b436a which is really not optimal. I don't even know, if commit cbdec69 is stable or not. Please, clarify.

Unfortunately, the new hardware requires this signed firmware files, so the faster incremental changes are really preferred to have something for test / update before users are disappointed.

Thank you.

lgirdwood, 2020-12-01:
@perexg apologies - there was some issues and delays fixing an issue that could have meant software changes outside of the FW/kernel between rc3 and now. This is something we need to improve upon.
The issue is now resolved and validation should complete for v1.6 final today at which point @mwasko can sign and release.
Fwiw, we are looking to overhaul the binary release infrastructure for v1.7 (due end January) and @marc-hb will be leading this effort. @marc-hb please align with @perexg on any requested requirements/improvements.

marc-hb, 2020-12-01:
> To avoid big git size, we always rebase the branch.

Binaries or not, force-pushing a branch with `stable` in the name is really not OK. Let's just not do that again.

git has unfortunately [no way to advertise volatile branches](https://public-inbox.org/git/xmqq60kc5iaj.fsf@gitster.mtv.corp.google.com/), just project-specific naming conventions. If we really need to force-push throw-away binaries we must at least use another branch with something "volatile-looking" in the name, say  a `v1.6-temp` branch for instance.

More widely shared Release Candidates must obviously never be deleted.

I'm well aware using git to release binaries is a major performance mistake but @xiulipan does `sof-bin` performance really matter? Is there some automation repeatedly cloning this that would be affected?

`linux-firmware` is full of binaries. No one pretends it's fast but I have not heard anyone complaining it's unusable either.

If yes git performance of `sof-bin` is a serious concern for some reason then an entire `sof-bin-temp` repo can provide a very simple and short term solution.

mwasko, 2020-12-02:
> > To avoid big git size, we always rebase the branch.
> 
> Binaries or not, force-pushing a branch with `stable` in the name is really not OK. Let's just not do that again.

I agree with that. Once we announce the release and create stable branch or any other target release branch we should stick to it and only add commits on top of that if necessary.

perexg, 2020-12-02:
> The issue is now resolved and validation should complete for v1.6 final today at which point @mwasko can sign and release.

Any update on this? I would like to package the final 1.6 ASAP. Thank you.

Also, the release scheme should be described in main https://github.com/thesofproject/sof-bin/blob/master/README.md file.

perexg, 2020-12-07:
> > The issue is now resolved and validation should complete for v1.6 final today at which point @mwasko can sign and release.
> 
> Any update on this? I would like to package the final 1.6 ASAP. Thank you.

Ping.

xiulipan, 2020-12-08:
@perexg sorry about the mess, I double confirmed the repo and find more issue with the v1.6-rc3 binaries. 
PS: we are now using PR mode for the stable-branch update. 
@lgirdwood Please help to check https://github.com/thesofproject/sof-bin/pull/26

> Also, the release scheme should be described in main https://github.com/thesofproject/sof-bin/blob/master/README.md file.

@lgirdwood @mengdonglin Any idea about the release scheme

lgirdwood, 2020-12-09:
@xiulipan https://github.com/thesofproject/sof-bin/pull/26 merged. The readme.md should contain a link to the latest binaries and some instructions on how to manually install them (these instruction should support copy and paste into a users terminal)..
@perexg apologies, one of our validation platforms and engineers has been impacted by a local lockdown and this has impacted our progress.

cinerea0, 2020-12-10:
To clarify, will future releases be tagged in their respective stable branches?

lgirdwood, 2020-12-10:
@cinerea0 yes - future releases will be tagged - the rebase was a mistake.

lgirdwood, 2020-12-18:
@perexg, @cinerea0  now released v1.6.1 - tagged as v1.6.1. Apologies, this has been slow to validate due to local lockdown restrictions for some of the validation team.

perexg, 2021-01-03:
It seems that the Jasper Lake firmware is missing in v1.6.1 (b77c851bc4ec1b6b552eaf1a61a66f3df4a13ab8). It is present in v1.6 branch (faa5bdc276651f3f8b7c77612fff770bd778987b).

mengdonglin, 2021-01-04:
@perexg @lgirdwood Jasper Lake firmware signed by Intel production key has not been verified on any product device yet, so we didn't include it in v1.6.1. Shall we only publish a Jasper Lake firmware signed by the community key?

lgirdwood, 2021-01-04:
@mengdonglin please do include in the release - I have confidence with CI and validation that it should be fine for JSL (based on the same IP in other platforms). We should still do the full validation when local Covid restrictions allow and can log/fix bugs in the mean time.

xiulipan, 2021-01-05:
@mengdonglin @lgirdwood add public signed JSL binary with https://github.com/thesofproject/sof-bin/pull/29

lgirdwood, 2021-01-05:
@marc-hb fyi we will need a ```make check``` rule similar to linux firmwares rule which must be run prior to publishing a release. This should check for missing files.

perexg, 2021-01-25:
Closing. All my reported issues were resolved.

closed 2021-01-25 by perexg