thesofproject/sof-bin #31: How to tell if SOF is working and has loaded the correct topology?

worldgeek, 2021-01-21:
I'm totally lost. Any help gratefully received.

go.sh ran successfully 
Firmware and topology appear to be loaded -  no errors appear in dmesg (alsa-info appended).
The system produces audio.

amixer does not appear to show any SOF components such as EQIIR.
I don't know which topology has been loaded or how to configure what should be loaded.

System
Up Extreme
Ubuntu 18.04.5 LTS
Kernel 5.10.0-rc5+ built according to SOF documentation 
sof-bin v1.6.1 downloaded from github
alsa - latest master from github as of November 2020

[alsa-info-sof.txt](https://github.com/thesofproject/sof-bin/files/5849686/alsa-info-sof.txt)

worldgeek, 2021-01-21:
My mistake!
I had not specified the `controls` parameter to amixer.

It would be nice though to know which topology is loaded.

Turning on [dynamic debug](https://thesofproject.github.io/latest/getting_started/intel_debug/suggestions.html#enable-dynamic-debug) shows that EQIIR is defined in whichever topology is loaded.
 `sof-audio-pci 0000:00:1f.3: tplg: ready widget id 11 pipe 2 type 33 name : EQIIR2.0 stream none`

`amixer -Dhw:0 controls | grep IIR`

`numid=34,iface=MIXER,name='EQIIR2.0 eqiir_bytes_2'`

juimonen, 2021-01-21:
dmesg | grep tplg
[    7.825066] sof-audio-pci 0000:00:0e.0: loading topology:intel/sof-tplg/sof-apl-pcm512x.tplg

I get this in my up2...

juimonen, 2021-01-21:
But you might need to enable the dynamic debug as described below:
https://thesofproject.github.io/latest/getting_started/intel_debug/suggestions.html

worldgeek, 2021-01-21:
Thanks for the tip on dynamic debug.
I see a different tplg file - does that matter?

[    4.525789] sof-audio-pci 0000:00:1f.3: loading topology:intel/sof-tplg/sof-hda-generic-4ch.tplg

worldgeek, 2021-01-21:
Unfortunately I still don't know if SOF is working correctly. 
Sorry for the newbie questions!

I'm running this test as suggested in the debugging link (it produces pink noise)
speaker-test -Dhw:0,0 -c2 -r48000 -f S16_LE

then loading different EQIIR coefficients
$ amixer -Dhw:0 controls | grep IIR
...
numid=34,iface=MIXER,name='EQIIR2.0 eqiir_bytes_2'

$ sof-bin/tools/v1.6.1/sof-ctl -Dhw:0 -n 34 -s sof/tools/ctl/eq_iir_bandpass.txt

but I can't hear any difference in the audio output 
What could be missing? Should I be reloading SOF after sof-ctl? How is that done?

I tried sof-bin/tools/v1.6.1/sof-logger -t -d sof-bin/lib/firmware/intel/sof/v1.6.1/sof-cnl-v1.6.1.ldc
but
error: Unable to open in file /sys/kernel/debug/sof/trace
(same error even when run with sudo)

worldgeek, 2021-01-21:
p.s.
Is this the right place for my questions? Would thesofproject/sof be better?
Thanks again.

juimonen, 2021-01-21:
for me this place is fine :)

I think the eq's in this topology are in the digital mic recording path... so if you don't have some extension board with dmics or the extreme doesn't have some preinstalled dmics -> hard to know the difference :)

intel/sof-tplg/sof-hda-generic-4ch.tplg means it is loading a topology with hda audio interface and 4 digital microphones... so it is somehow detecting this from the bios. Unfortunately I've not worked with up extreme, I have the older version myself.

there should be python tool sof-tplgreader (or similar name) that you could run on the tplg binary and get a png image of the topology -> much better to understand what component is where

plbossart, 2021-01-21:
@worldgeek I think there's a mistake in the UpExtreme BIOS with the default HDaudio BIOS setting, as @juimonen you don't have any microphones attached and hence the only capture path that's available to you is the headset input with the 3.5mm jack.

you can make this go away by adding this to /etc/modprobe.d/alsa-base.conf

options snd_sof_intel_hda_common dmic_num=0

worldgeek, 2021-01-23:
Thanks for the suggestions which I hope I've followed correctly.
Looks like there's something else missing which is causing a problem. Maybe it's a magical BIOS setting.

Screenshots of BIOS settings for HDaudio and DSP are attached with alsa-info output.

Added options snd_sof_intel_hda_common dmic_num=0 >> /etc/modprobe.d/alsa-base.conf

Upgraded to BIOS 1.8 from https://downloads.up-community.org/download/up-xtreme-uefi-bios-v1-8/
There were a couple of gotchas
1. Booting into BIOS while pressing DEL has a different meaning than pressing F7, only F7 gets the UEFI menu
2. Password upassw0rd must be entered to reach adminstrator mode and the CRB Setup menu

dmesg now shows
sof-audio-pci 0000:00:1f.3: request_firmware intel/sof/sof-cnl.ri successful
...
sof-audio-pci 0000:00:1f.3: loading topology:intel/sof-tplg/sof-hda-generic.tplg

speaker-test -Dhw:0,0 -c2 -r48000 -f S16_LE
Makes a thump sound then silence and then another thump a few seconds after it is stopped

aplay sof/alsa-utils/speaker-test/samples/Front_Left.wav
Makes a thump sound, then silence and then another thump after a delay

amixer -Dhw:0 controls | grep IIR
numid=34,iface=MIXER,name='EQIIR2.0 eqiir_bytes_2'
numid=37,iface=MIXER,name='EQIIR4.0 eqiir_bytes_4'

[image]
[image]
[alsa-info-up-extreme-bios-1.8.txt](https://github.com/thesofproject/sof-bin/files/5860471/alsa-info-up-extreme-bios-1.8.txt)

plbossart, 2021-01-25:
@worldgeek have you played with alsamixer and codec settings? This should work out of the box with PulseAudio and UCM. Not sure if at the lowest level you have configured the codec correctly.

worldgeek, 2021-01-25:
@plbossart, thanks.  I did try changing two BIOS settings Advanced Link Audio Mode and DMIC but there was no improvement.
The settings shown are the BIOS defaults.

Any suggestions for codec configuration? In the BIOS or elsewhere?
How can I check UCM? 

Alasamixer shows 

> Card: sof-hda-dsp
> Chip: Realtek ALC892

What should I be looking for?

Would we expect to see a different toplology than `intel/sof-tplg/sof-hda-generic.tplg`?

Running alsa-info shows errors
> 
> cat: '/sys/module/snd_soc_skl_hda_dsp/parameters/*': No such file or directory
> alsactl: get_control:256: Cannot read control '3,0,0,Playback Channel Map,0': Invalid argument
> cat: /tmp/alsa-info.L7Kf28otjF/alsactl.tmp: No such file or directory

I am new to this software stack and have never seen/heard  it working so I don't know what to expect.
It would be helpful to see the settings of a working installation on UP Xtreme,

[alsa-info-up-xtreme.txt](https://github.com/thesofproject/sof-bin/files/5868121/alsa-info-up-xtreme.txt)

worldgeek, 2021-01-25:
Thanks for all the help so far. There is now audio output (again).
Sadly I _may_ have broken the setup when adding a USB headset to the KVM switch I'm using with the UP board.
Sorry for the user error.

alsamixer did not show anything as muted but pavucontrol did.
I'd also assumed that hearing speaker thumps meant that the audio was active.

Followed this sequence:
$ speaker-test -Dhw:0,0 -c2 -r48000 -f S16_LE
makes the bump sound
$ pavucontrol
_unmute_ sof-hda-dsp Output Device
(speaker-test can't run if pavucontrol is running)
quit pavucontrol

$ speaker-test -Dhw:0,0 -c2 -r48000 -f S16_LE
Produces pink noise as expected
kernel messages from sof-audio-pci seem reasonable

So it's back to trying sof-ctl again as mentioned earlier in this thread.
Setting parameters on EQIIR appears to do nothing (following instructions here https://thesofproject.github.io/latest/developer_guides/tuning/sof-ctl.html). Is this because of the topology set by sof-hda-generic.tplg?

Is there anything else that can be tried which would demonstrate sof-ctl tuning a dynamic parameter?
Should I expect to see any additional controls in alsamixer?

worldgeek, 2021-01-25:
P.S. I've been assuming that the analog audio jack on the UP board corresponds to "Passthrough Playback". And so it should react to setting IIR coeffs on amixer control corresponding to EQIIR2.0?

> amixer -Dhw:0 controls | grep EQ
> numid=34,iface=MIXER,name='EQIIR2.0 eqiir_bytes_2'
> numid=37,iface=MIXER,name='EQIIR4.0 eqiir_bytes_4'

> $ ./work/sof-test/tools/sof-tplgreader.py /lib/firmware/intel/sof-tplg/sof-hda-generic.tplg 
> pcm=HDA Analog;id=0;type=playback;cap_name=Passthrough Playback 0;pga=PGA1.0;eq=EQIIR2.0;fmts=S16_LE S24_LE;fmt=S16_LE;rate_min=48000;rate_max=48000;ch_min=2;ch_max=2;rate=48000;channel=2;dev=hw:0,0;snd=/dev/snd/pcmC0D0p;
> pcm=HDA Analog;id=0;type=capture;cap_name=Highpass Capture 0;pga=PGA2.0;eq=EQIIR2.0;fmts=S16_LE S24_LE;fmt=S16_LE;rate_min=48000;rate_max=48000;ch_min=2;ch_max=2;rate=48000;channel=2;dev=hw:0,0;snd=/dev/snd/pcmC0D0c;
> pcm=HDA Digital;id=1;type=playback;cap_name=Passthrough Playback 1;pga=PGA3.0;eq=EQIIR4.0;fmts=S16_LE S24_LE;fmt=S16_LE;rate_min=48000;rate_max=48000;ch_min=2;ch_max=2;rate=48000;channel=2;dev=hw:0,1;snd=/dev/snd/pcmC0D1p;
> pcm=HDA Digital;id=1;type=capture;cap_name=Highpass Capture 1;pga=PGA4.0;eq=EQIIR4.0;fmts=S16_LE S24_LE;fmt=S16_LE;rate_min=48000;rate_max=48000;ch_min=2;ch_max=2;rate=48000;channel=2;dev=hw:0,1;snd=/dev/snd/pcmC0D1c;
> pcm=HDMI1;id=3;type=playback;cap_name=Passthrough Playback 3;pga=PGA7.0;fmts=S16_LE S24_LE;fmt=S16_LE;rate_min=48000;rate_max=48000;ch_min=2;ch_max=2;rate=48000;channel=2;dev=hw:0,3;snd=/dev/snd/pcmC0D3p;
> pcm=HDMI2;id=4;type=playback;cap_name=Passthrough Playback 4;pga=PGA8.0;fmts=S16_LE S24_LE;fmt=S16_LE;rate_min=48000;rate_max=48000;ch_min=2;ch_max=2;rate=48000;channel=2;dev=hw:0,4;snd=/dev/snd/pcmC0D4p;
> pcm=HDMI3;id=5;type=playback;cap_name=Passthrough Playback 5;pga=PGA9.0;fmts=S16_LE S24_LE;fmt=S16_LE;rate_min=48000;rate_max=48000;ch_min=2;ch_max=2;rate=48000;channel=2;dev=hw:0,5;snd=/dev/snd/pcmC0D5p;
>

plbossart, 2021-01-25:
@worldgeek you can do the same as pulseaudio with an 'alsaucm' command. 

alsaucm -c sof-hda-dsp set _verb HiFi set _enadev Headphone

the card name and/or Headphone/Headphones might need some tweaking, but the point is you need mixer settings prior to playback/capture.

@singalsu can you help with the EQ use?

worldgeek, 2021-01-26:
Thanks for the continued support everyone :-)

It seems I don't have alsa ucm2, despite having updated alsa.
I see a git repo for ucm but there are no installation instructions.
Any clues?

`alsaucm -c sof-hda-dsp set _verb HiFi set _enadev Headphone`

> ALSA lib parser.c:2179:(load_toplevel_config) Unable to find the top-level configuration file '/usr/share/alsa/ucm2/ucm.conf'.
> ALSA lib main.c:1014:(snd_use_case_mgr_open) error: failed to import sof-hda-dsp use case configuration -2
> alsaucm: error failed to open sound card sof-hda-dsp: No such file or directory

As mentioned earlier in the thread, _alsamixer_ does know about sof-hda-dsp.

plbossart, 2021-01-26:
@worldgeek you need to update alsa-ucm-conf, it's a different package. I remove the default in /usr/share/alsa/ucm2 and replace it with the latest from https://github.com/alsa-project/alsa-ucm-conf.git

You will need anyways to play with UCM files yourself to update the processing and filter configurations.

It's hard to keep track of all the steps, it would be really appreciated if you can keep a list of everything we missed in sof-docs so that we can do a better job for the next users :-)

plbossart, 2021-01-26:
oh and in terms of install instructions it's a simple copy that I do. We are not that skilled and just want to get the job done on our development machines.

cd alsa-ucm-conf; cp -rf ucm2 /usr/share/alsa/

worldgeek, 2021-01-27:
@plbossart yes I'll be writing up my findings.

alsaucm -c sof-hda-dsp set _verb HiFi set _enadev Headphones
Gets audio output working e.g. speaker-test.

I'm lacking any knowledge of the expected user experience here.
Should something appear in alsamixer which refers to that alsaucm setting?
Or is there a pulseaudio tool?
What configuration is needed to "turn on" EQIIR2 for the Headphones output?

Looking at those ucm files I have no idea how to configure them for the goal of hearing the effect of sof-ctl setting a different EQ.
@singalsu, can you suggest something?

I'm happy to follow any documentation but haven't been able to find anything useful as yet. Not even sure what to search for :-(
Are there any user-level docs which describe how a normal user would tweak EQ settings on their Linux laptop running SOF?

worldgeek, 2021-01-27:
It would be very helpful to see what a typical setup _should_ look/sound like.

plbossart, 2021-01-27:
@worldgeek there's two levels for testing.

a) alsaucm + speaker-test: this is for developers who want to bypass PulseAudio and all alsa-lib layers. This is typically what we use for SOF development and validation.

b) pulseaudio. In this case the UCM file is used by pulseaudio, and sinks/sources are enabled automagically. You should see in the Settings/Sound UI a change when you plug/unplug your headset (that may or may not work depending on codec configurations, it may be that on this board it exposes a 'speaker' even when there's none...).

For the EQ control, there's no UI I am aware of. There is a sof-ctl application but Seppo Ingalsuo @singalsu is our resident DSP guru. You can also embed the coefficients in a topology file or in a UCM file. For now you should plan for 'headset' and 'speaker' settings that are applied when the relevant sink/source is applied.

singalsu, 2021-01-27:
@worldgeek Nice that you are looking for EQs! The default topology sof-apl-pcm512x.tplg (seems UP extrame uses the same as UP2 that I have, right?). There's a tool in sof-test repo tplgtool.py that creates png format plots of .tplg files:

[image]

If you need an EQ to playback pipeline the loaded topology needs to be replaced by another. There's one example that is not part of the binary v1.6.x release but should still work sof-apl-eq-pcm512x.tplg. It's built by command scripts/build_tools.sh

[image]

But the topology misses functionality that you will likely need (HDMI). So a version of sof-apl-pcm512x.m4 should be created where pipe-volume-playback.m4 macro is replaced one of

- pipe-eq-iir-volume-playback.m4
- pipe-eq-fir-volume-playback.m4
- pipe-eq-iir-eq-fir-volume-playback.m4

With EQ present in the the loaded topology, the response can be changed via the embedded coefficients or via sof-ctl. For some topologies the filter coefficients m4 blob name can be passed from top level topology CMakelist.txt. In runtime the control name can be looked up with command `amixer controls | grep EQ`. The control number needs lest typing.

Then the control can be retrieved from driver with `sof-ctl -n <number>` . The control (filter coefficients) can be set with data created by EQ tool with `sof-ctl -n <number> -s <filename>` . If you need different speaker and headphone EQs with the same pipeline then the response updated like sof-ctl does.

There may be later in SOF a EQ preset capibility so the response could be easily switched via alsamixer with pre-configured coefficients.

singalsu, 2021-01-27:
... continued. I just booted my UP2 with sof-apl-eq-pcm512x.tplg copied over sof-apl-pcm512x.tplg. Now I see the controls

```
$ amixer controls | grep EQ
numid=24,iface=MIXER,name='EQFIR1.0 eqfir_coef_1'
numid=23,iface=MIXER,name='EQIIR1.0 eqiir_coef_1'
```

Get the current configuration:

```
$ sof-ctl -n 23
Control size is 1024.
Retrieving configuration for device hw:0 control numid=23.
Success.
hdr: magic 0x00464f53
hdr: type 0
hdr: size 88 bytes
hdr: abi 3:18:0
4607827,0,88,50405376,0,0,0,0,88,2,1,0,0,0,0,4294967295,4294967295,1,1,0,0,0,0,0,0,0,0,538145694,0,32690

$ sof-ctl -n 24
Control size is 4096.
Retrieving configuration for device hw:0 control numid=24.
Success.
hdr: magic 0x00464f53
hdr: type 0
hdr: size 56 bytes
hdr: abi 3:18:0
4607827,0,56,50405376,0,0,0,0,56,65538,0,0,0,0,4294967295,4294901764,0,0,0,0,16384,0

```

Set new filter coefficients, these are examples generated in tools/tune/eq:

```
$ sof-ctl -n 23 -s eq/eq_iir_loudness.txt 
Control size is 1024.
Applying configuration "eq/eq_iir_loudness.txt" into device hw:0 control numid=23.
4607827,0,204,50331648,0,0,0,0,204,4,2,0,0,0,0,1,1,1,1,1,1,0,0,0,0,0,0,0,0,538145694,0,32690,3,3,0,0,0,0,3237960220,2130690484,297056159,3694463533,303476880,0,16384,3302357025,2064935920,245217319,3776455865,274003881,0,16384,4106268671,4130185751,69182517,4179658584,365641401,4294967292,25468
Success.
hdr: magic 0x00464f53
hdr: type 0
hdr: size 204 bytes
hdr: abi 3:0:0
4607827,0,204,50331648,0,0,0,0,204,4,2,0,0,0,0,1,1,1,1,1,1,0,0,0,0,0,0,0,0,538145694,0,32690,3,3,0,0,0,0,3237960220,2130690484,297056159,3694463533,303476880,0,16384,3302357025,2064935920,245217319,3776455865,274003881,0,16384,4106268671,4130185751,69182517,4179658584,365641401,4294967292,25468

singalsu@piikkisika:~$ sof-ctl -n 24 -s eq/eq_fir_mid.txt 
Control size is 4096.
Applying configuration "eq/eq_fir_mid.txt" into device hw:0 control numid=24.
4607827,0,128,50380800,0,0,0,0,128,65538,0,0,0,0,0,65576,0,0,0,0,1202154746,4222755303,3873958283,4024953600,4126012299,4176934804,4217764409,4250664050,4273864222,4288806716,2490349,6750285,8126582,7733371,6488174,4980824,3473472,2293804,1245211,1245197
Success.
hdr: magic 0x00464f53
hdr: type 0
hdr: size 128 bytes
hdr: abi 3:12:0
4607827,0,128,50380800,0,0,0,0,128,65538,0,0,0,0,0,65576,0,0,0,0,1202154746,4222755303,3873958283,4024953600,4126012299,4176934804,4217764409,4250664050,4273864222,4288806716,2490349,6750285,8126582,7733371,6488174,4980824,3473472,2293804,1245211,1245197

```

Hope this explained the usage. For doing EQ design with Octave or Matlab look example in tune: example_iir_eq.m example_spk_eq.m, etc. There's also a gui.m for interactive graphical EQ design and run-time update to device via ssh but it's much simplified from the script/command line EQ calculate procedure.

worldgeek, 2021-01-27:
@singalsu I'm excited to try your suggestions! Things are slowly starting to make more sense :)

A few points which I'm unclear on (I'm a beginner with Linux audio / SOF etc)

1. UP extreme is choosing sof-hda-generic.tplg, will that cause any problems if overwritten with sof-apl-eq-pcm512x.tplg?
At present I don't quite understand the relationship between a topology and the actual hardware.
(I will take your excellent advice to blacklist snd_sof_pci).

2.` ./sof-test/tools/sof-tplgreader.py /lib/firmware/intel/sof-tplg/sof-hda-generic.tplg` shows:
>  pcm=HDA Analog;id=0;type=playback;cap_name=Passthrough Playback 0;pga=PGA1.0;eq=EQIIR2.0;...channel=2;dev=hw:0,0;snd=/dev/snd/pcmC0D0p

However tplgtool shows

> pcm=HDA Analog;id=0;type=both;fmt=S16_LE;rate_min=48000;rate_max=48000;ch_min=2;ch_max=2;
and the graph shows EQIIR only on inputs.

What does the "eq=EQIIR2.0" from sof-tplgreader.py mean?

3. My setup only has a headphone output, no HDMI audio - does this alter what should be in the topology?

Thanks

lgirdwood, 2021-01-27:
> 1. UP extreme is choosing sof-hda-generic.tplg, will that cause any problems if overwritten with sof-apl-eq-pcm512x.tplg?
>    At present I don't quite understand the relationship between a topology and the actual hardware.

I can answer this part. The topology define audio pipelines AND how they connect to physical audio hardware endpoints. i.e. ALSA PCM:0,0 is route via pipeline X with processing, Y and Z and rendered on HDMI device 0.

So the HW enpoints cant be changed as they are part of your board, but the pipeline and components that connect to the HW endpoint can be anything you like providing the DSP and FW have the resources.

worldgeek, 2021-01-27:
The support here is outstanding, many thanks once again.
OK. So I must make a topology _like_ sof-apl-eq-pcm512x.tplg which is compatible with the setup in sof-hda-generic.tplg.
Trying to copy sof-apl-eq-pcm512x over sof-hda-generic showed some errors. Which I think is exactly what @lgirdwood was warning about - so it's a beautiful illustration ;-)

> 
> [70883.991003] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x30010000 size 20
> [70883.991014] sof-audio-pci 0000:00:1f.3: error: DSP failed to add widget id 0 type 27 name : SSP5.OUT stream SSP5-Codec reply 0
> [70883.991021] sof-audio-pci 0000:00:1f.3: ASoC: failed to load widget SSP5.OUT
> [70883.991025] sof-audio-pci 0000:00:1f.3: ASoC: topology: could not load header: -22
> [70883.991036] sof-audio-pci 0000:00:1f.3: error: tplg component load failed -22
> [70883.991049] sof-audio-pci 0000:00:1f.3: error: failed to load DSP topology -22
> [70883.991055] sof-audio-pci 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
> [70883.991060] sof-audio-pci 0000:00:1f.3: ASoC: failed to probe component -22
> [70883.991089] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: failed to instantiate card -22

singalsu, 2021-01-27:
@worldgeek Sorry, I think I've misunderstood (not familiar with this device). You should then use a custom sof-hda-generic topology if that's what kernel by default uses.  There's in binary release sof-hda-generic-eq-2ch.tplg, copy it over sof-hda-generic-2ch.tplg in /lib/firmware/intel/sof-tplg. Here's graph of the topology:

[image]

Amixer should find the EQIIR1.0 and EQFIR1.0 instances those you can use.  Run-time reconfiguration is simple with sof-ctl. Customization of sof-hda-generic is more cumbersome m4 editing since the headphone/spekaer endpoint filter coefficients are not propagated to toplevel CMakeList.txt (yet). So editing those would need add of PIPELINE_FILTER1 and PIPELINE_FILTER2 definitions to sof-hda-generic just before call of pipe-xxx-playback.m4 variant used.

I'll help you further with that if you want to experiment.

singalsu, 2021-01-27:
@worldgeek If your device is without built-in DMICs (if plain sof-hda-generic.tplg is used). In that case copy topology sof-hda-generic-eq-2ch.tplg over topology sof-hda-generic.tplg. The topology that I proposed adds those endpoints but that should not prevent usage, only creates those for-you-not-working sound devices. A custom sof-hda-generic-eq.tplg without DMIC endpoints can be built by a simple edit to topology CMakeList.txt.

Find from there line

`"sof-hda-generic\;sof-hda-generic-eq-2ch\;-DCHANNELS=2\;-DPPROC=eq-iir-eq-fir-volume\;-DDMICPROC_FILTER1=eq_iir_coef_highpass_40hz_20db_48khz.m4\;-DDMIC16KPROC_FILTER1=eq_iir_coef_highpass_40hz_20db_16khz.m4"`

Make a copy of the line with target `sof-hda-generic-eq` and set there `CHANNELS=0`.

singalsu, 2021-01-27:
@worldgeek Not my day... I found now your topology from middle of this discussion, it's sof-hda-generic-4ch.tplg. Forget the above and copy `sof-hda-generic-eq-4ch.tplg` over `sof-hda-generic-4ch.tplg`.

`[ 4.525789] sof-audio-pci 0000:00:1f.3: loading topology:intel/sof-tplg/sof-hda-generic-4ch.tplg`

worldgeek, 2021-01-27:
@singalsu Thanks. I was previously told not to use topology:intel/sof-tplg/sof-hda-generic-4ch.tplg. Possibly at that time it was understood I wanted to change the input EQ rather than the output.
From @plbossart :

> @worldgeek I think there's a mistake in the UpExtreme BIOS with the default HDaudio BIOS setting, as @juimonen you don't have any microphones attached and hence the only capture path that's [...]

I'll revert this line and try again.

In the meantime I had attempted to create sof-hda-generic-eq myself and "almost" got the right line in CMakeLists.txt
`"sof-hda-generic\;sof-hda-generic-eq\;-DCHANNELS=0\;-DPPROC=eq-iir-eq-fir-volume"`
but when this topology is used (image below)  a message appears in the kernel log

> [ 2396.031938] skl_hda_dsp_generic skl_hda_dsp_generic: control 2:0:0:PGA1.0 1 Master Playback Volume:0 is already present
> [ 2396.031949] sof-audio-pci 0000:00:1f.3: ASoC: failed to add widget PGA1.0 dapm kcontrol PGA1.0 1 Master Playback Volume: -16

[image]

worldgeek, 2021-01-27:
With topology:intel/sof-tplg/sof-hda-generic-4ch.tplg I hear nothing through the headphone output (and as mentioned I have no other way to hear audio).

I then tried
`alsaucm -c sof-hda-dsp set _verb HiFi set _enadev Headphones
`
Again no effect.

Ran pavucontrol. This shows the headphones connected as input but not as output.

So I've gone back to alsa-base.conf "options snd_sof_intel_hda_common dmic_num=0".
And your suggestion of how to make sof-hda-generic.tplg

worldgeek, 2021-01-27:
This version of sof-hda-generic-eq produces some errors and no audio output.

> 	"sof-hda-generic\;sof-hda-generic-eq\;-DCHANNELS=0\;-DPPROC=eq-iir-eq-fir-volume\;-DDMICPROC_FILTER1=eq_iir_coef_highpass_40hz_20db_48khz.m4\;-DDMIC16KPROC_FILTER1=eq_iir_coef_highpass_40hz_20db_16khz.m4"

Selected messages

> [    4.551451] sof-audio-pci 0000:00:1f.3: warning: neither Linked source component codec0_in nor sink component HDA1.IN is of buffer type, ignoring link
> [    4.551458] sof-audio-pci 0000:00:1f.3: warning: neither Linked source component codec1_in nor sink component HDA3.IN is of buffer type, ignoring link
> [    4.551473] sof-audio-pci 0000:00:1f.3: warning: neither Linked source component codec2_in nor sink component HDA3.IN is of buffer type, ignoring link
> [    4.585857] sof-audio-pci 0000:00:1f.3: ASoC: failed to add widget PGA1.0 dapm kcontrol PGA1.0 1 Master Playback Volume: -16
> 
And no output devices are shown in avucontrol.

worldgeek, 2021-01-27:
On reverting to sof-bin v1.6.1 and its sof-hda-generic.tplg there are no output devices shown by pavucontrol.
<sad face>
alsa-info attached.

[alsa-info-up-extreme-no-devices.txt](https://github.com/thesofproject/sof-bin/files/5882667/alsa-info-up-extreme-no-devices.txt)

juimonen, 2021-01-27:
@worldgeek me and @singalsuo will try to generate you some eq pipeline tomorrow for hda topology. Sorry we've been running different topologies in older up2... maybe easier we try to generate you something -> you can go on there... it is unfortunately pretty hairy to modify the topologies if you haven't done that before.

worldgeek, 2021-01-27:
@juimonen that would be fantastic. You guys are great.
I quite understand that most of you don't run this UP Extreme hardware so it's making remote debugging very tricky.

And I discovered that audio from the Up Extreme fails (other than "thumps") when my expensive portable speaker (Bose) is charging from a USB port on my Macbook.  Seems to be partially related to the UP board since the same setup is OK with other audio devices.
OMG!
So the audio failure and _sad face_ referred to earlier tonight was actually caused by charging the speaker which happened to run out of power just as I was switching topologies.

Still no devices shown in pavucontrol. Will check after a reboot.

singalsu, 2021-01-28:
I'll today make a PR to enhance build of sof-hda-generic topologies with EQs. It will be much easier to work with that (I hope)!

singalsu, 2021-01-28:
@worldgeek Please check https://github.com/thesofproject/sof/pull/3796 . It's still draft, I'll turn it into proposal once CI shows green.

When doing comparison (`diff -udr topology.old topology`) of ALSA topology .conf files before and after the PR I noticed that there was a change in HDA capture EQ filter coefficients for sof-hda-generic-eq*.conf. It was because previous version of topology defined PIPELINE_FILTER1 and continued without undefine. It replaced in those topologies the intended high-pass filter (to mitigate codec capture settling ) with pass-trough. The mistake was before unnoticed. So as takeaway, it's good idea to browse trough the .conf for the topology you that was edited. This time I was lucky to fix a bug while doing this improvement.

worldgeek, 2021-01-28:
@singalsu Hurrah! That worked. 🎉 Setting IIR and FIR parameters with sof-ctl now make a definite difference to the sound output on the speaker. 🤓

singalsu, 2021-01-29:
@lgirdwood Need to get the sof-hda-generic-eq.tplg included to v1.7. The SOF PR also fixes the HDA capture EQ response for sof-hda-generic-eq-1/2/3/4ch.tplg to the proper high-pass that is used in no playback EQ -versions of sof-hda-generic.tplg.

lgirdwood, 2021-01-29:
@singalsu I assume you mean https://github.com/thesofproject/sof/pull/3796 now merged ?

singalsu, 2021-01-29:
@lgirdwood Yep, that one.

worldgeek, 2021-02-03:
I noticed @singalsu's patch in master. We have been building from origin/stable-v1.6 and applied the patch manually (git apply ...), should we change branches in anticipation of v1.7?

My colleague is checking over the notes about setting up the UP Extreme for SOF which I prepared from all the help received here. We expect to share with the SOF project soon.

lgirdwood, 2021-02-04:
@worldgeek v1.7-rc1 now tagged.

worldgeek, 2021-02-04:
I forgot to mention that the pulseaudio pavucontrol issue referred to above (not showing sof devices) was "fixed" by removing SectionDevice."Mic1"  from  /usr/share/alsa/ucm2/sof-hda-dsp/Hifi.conf (to match `options snd_sof_intel_hda_common dmic_num=0` in /etc/modprobe.d/alsa-base.conf).

marc-hb, 2021-10-05:
> Hurrah! That worked

Can we close this?