thesofproject/sof-bin #32: Cannot load community signed firmware v1.6.1 on Up Extreme

worldgeek, 2021-01-21:
According to what I've read, the UP series of boards are able to use a community key for firmware.
This probem might be nothing to do with signed firmware but I'd like to know how to solve it.

My own newly built firmware fails to load so I tried sof-bin v1.6.1 from github.
Ran go.sh to install to /lib/firmware.

Then
`sudo ln -s /lib/firmware/intel/sof/community/sof-cnl.ri /lib/firmware/intel/sof`

On booting the firmware is apparently read but cl_copy_fw fails (dmesg log attached).
```
sof-audio-pci 0000:00:1f.3: request_firmware intel/sof/sof-cnl.ri successful
...
sof-audio-pci 0000:00:1f.3: error: cl_copy_fw: timeout HDA_DSP_SRAM_REG_ROM_STATUS 
```

System
Up Extreme
Ubuntu 18.04.5 LTS
Kernel 5.10.0-rc5+ built according to SOF documentation
sof-bin v1.6.1 downloaded from github
alsa - latest master from github as of November 2020

[dmesg-sof-community.txt](https://github.com/thesofproject/sof-bin/files/5850017/dmesg-sof-community.txt)
[alsa-info-sof-community.txt](https://github.com/thesofproject/sof-bin/files/5850018/alsa-info-sof-community.txt)

juimonen, 2021-01-21:
@plbossart FYI do we have the community key in up2 extreme set up? I don't have that board myself..

lgirdwood, 2021-01-22:
@plbossart @kv2019i fyi here's the dmesg - but it looks like a FW load issue and I'm not sure we are getting to FW signature verification. 
@worldgeek I dont have this board, but IIRC the older UP^2 required the DSP to be enabled in the BIOS menu. Can you chekc the BIOS audio config and make sure DSP is enabled. 
```
[    3.844869] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0xc00
[    3.844875] sof-audio-pci 0000:00:1f.3: found ML capability at 0xc00
[    3.844877] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x800
[    3.844881] sof-audio-pci 0000:00:1f.3: found DSP capability at 0x800
[    3.844882] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x500
[    3.844886] sof-audio-pci 0000:00:1f.3: found GTS capability at 0x500
[    3.844888] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x1f00
[    3.844891] sof-audio-pci 0000:00:1f.3: found DRSM capability at 0x1f00
[    3.844893] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x700
[    3.844897] sof-audio-pci 0000:00:1f.3: found SPIB capability at 0x700
[    3.844930] sof-audio-pci 0000:00:1f.3: DMA mask is 64 bit
[    3.844936] sof-audio-pci 0000:00:1f.3: hda global caps = 0x1009701
[    3.844938] sof-audio-pci 0000:00:1f.3: detected 9 playback and 7 capture streams
[    3.845064] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    3.845066] sof-audio-pci 0000:00:1f.3: using IPC IRQ 133
[    3.874903] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[    3.874907] sof-audio-pci 0000:00:1f.3: PP capability, will probe DSP later.
[    3.877860] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000040 at  00000048
[    3.877865] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
[    3.883112] sof-audio-pci 0000:00:1f.3: codec_mask = 0x5
[    3.883124] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[    3.883127] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000040 at  00000048
[    3.885346] acpi PRP00001:00: ACPI reports 4 SDW Link devices
[    3.885351] acpi PRP00001:00: Link 0 not selected in firmware
[    3.885353] acpi PRP00001:00: Link 1 not selected in firmware
[    3.885354] acpi PRP00001:00: Link 2 not selected in firmware
[    3.885449] acpi PRP00001:00: Link 3 not selected in firmware
[    3.885451] sof-audio-pci 0000:00:1f.3: skipping SoundWire, no links enabled
[    3.885525] sof-audio-pci 0000:00:1f.3: HDA codec #0 probed OK: response: 10ec0892
[    3.886521] hdaudio ehdaudio0D0: loading codec module: hdaudio:v10EC0892r00100302a01
               
[    3.904022] sof-audio-pci 0000:00:1f.3: HDA codec #2 probed OK: response: 8086280b
[    3.904600] hdaudio ehdaudio0D2: loading codec module: hdaudio:v8086280Br00100000a01
               
[    3.914201] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    3.914204] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    3.914215] sof-audio-pci 0000:00:1f.3: DMICs detected in NHLT tables: 4
[    3.914247] sof-audio-pci 0000:00:1f.3: loading firmware
[    3.919937] sof-audio-pci 0000:00:1f.3: request_firmware intel/sof/sof-cnl.ri successful
[    3.919940] sof-audio-pci 0000:00:1f.3: firmware doesn't contain extended manifest
[    3.919953] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000002 at  00000048
[    3.919957] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
[    3.919964] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[    3.919966] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000004 at  00000044
[    3.919967] sof-audio-pci 0000:00:1f.3: booting DSP firmware
[    3.920008] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[    3.920022] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[    3.920024] sof-audio-pci 0000:00:1f.3: period_bytes:0x0
[    3.920025] sof-audio-pci 0000:00:1f.3: periods:1
[    3.920066] sof-audio-pci 0000:00:1f.3: Attempting iteration 0 of Core En/ROM load...
[    3.920072] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
[    3.920590] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0f successful
[    3.920594] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f00 successful
[    3.920596] sof-audio-pci 0000:00:1f.3: unstall/run core: core_mask = f
[    3.920599] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask f
[    3.920605] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0000 successful
[    3.920606] sof-audio-pci 0000:00:1f.3: unstall/run core: core_mask = 1
[    3.920609] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
[    3.923855] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x80000000 successful
[    3.923865] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0e0e successful
[    3.923869] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf010e0e successful
[    3.923871] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask e
[    3.926741] random: crng init done
[    3.926743] random: 7 urandom warning(s) missed due to ratelimiting
[    3.991886] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x5000001 successful
[    4.000163] systemd-udevd (256) used greatest stack depth: 12424 bytes left
[    6.992363] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x80000012 timedout
[    6.992368] sof-audio-pci 0000:00:1f.3: error: cl_copy_fw: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
[    6.992379] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x140000 successful
[... cut ...]
[    6.992445] sof-audio-pci 0000:00:1f.3: error: failed to reset DSP
[    6.992448] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware -110
[    7.048370] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1c003c timedout
[    7.048375] sof-audio-pci 0000:00:1f.3: error: hda_dsp_core_reset_enter: timeout on HDA_DSP_REG_ADSPCS read
[    7.048381] sof-audio-pci 0000:00:1f.3: error: dsp core reset failed: core_mask 1
[    7.048678] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -110
```

juimonen, 2021-01-22:
@lgirdwood I think @worldgeek could load some firmware already...? like here: https://github.com/thesofproject/sof-bin/issues/31

plbossart, 2021-01-22:
@worldgeek @lgirdwood If you want to use the community key firmware on Up Extreme, we are missing a quirk to modify the path. I forgot to add it to GitHub, it was on my work branch.

But with the updated BIOS you can use the signed firmware as well. Both keys are supported, so I am not sure what is happening here.

lgirdwood, 2021-01-22:
@plbossart Btw, what version of the updated BIOS does @worldgeek need to install ?

plbossart, 2021-01-22:
https://downloads.up-community.org/download/up-xtreme-uefi-bios-v1-8/

worldgeek, 2021-01-23:
After upgrading the bios to 1.8 there is no change when attempting to use the public-signed firmware v1.6.1.

sof-audio-pci 0000:00:1f.3: error: cl_copy_fw: timeout HDA_DSP_SRAM_REG_ROM_STATUS read

As tracked on #31, at present we seem able to load the Intel signed firmware but it doesn't seem to be functioning correctly.

worldgeek, 2021-02-03:
Thanks for all the help with #31 :-)
Pinging this issue because we need the ability to install our own code in the DSP firmware.

plbossart, 2021-02-03:
@worldgeek can you move all the existing directories for /lib/firmware/intel/sof and sof-tplg, and re-create one where you have the 1.6.1 firmware and topology? There are multiple cases where the Ubuntu installs conflict with sof-bin packages.

lgirdwood, 2021-02-04:
@worldgeek any update today with the directory fix above ?

worldgeek, 2021-02-04:
@plbossart, @lgirdwood 
Sorry guys, late seeing your messages since I wasn't working earlier today.

The firmware is installed by running sudo ./go.sh in sof-bin which removes all of sof and sof-tplg from /lb/firmware/intel - is this what you mean?

To choose the community signed firmware I linked sof-cnl.ri from sof/community. 

> sudo rm /lib/firmware/intel/sof/sof-cnl.ri
> sudo ln -s /lib/firmware/intel/sof/community/sof-cnl.ri /lib/firmware/intel/sof
> uname -a
>  "Linux up 5.4.0-1-generic #0~upboard2-Ubuntu SMP Sun Oct 25 14:06:23 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux"

Note that in the first report for this issue I had a different kernel - 5.10.0-rc5+ built according to SOF documentation, whereas here it's the recommended UP kernel from repository ppa:aaeon-cm/5.4-upboard.

dmesg shows

> [    3.520723] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
> [    3.529338] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
> [    3.529341] sof-audio-pci 0000:00:1f.3: PCI DSP detected
> [    3.529355] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
> [    3.529594] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
> [    3.529602] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
> [    4.317506] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
> [    4.372019] sof-audio-pci 0000:00:1f.3: Turning i915 HDAC power on
> [    4.376892] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0xc00
> [    4.376895] sof-audio-pci 0000:00:1f.3: found ML capability at 0xc00
> [    4.376898] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x800
> [    4.376899] sof-audio-pci 0000:00:1f.3: found DSP capability at 0x800
> [    4.376902] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x500
> [    4.376903] sof-audio-pci 0000:00:1f.3: found GTS capability at 0x500
> [    4.376906] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x1f00
> [    4.376907] sof-audio-pci 0000:00:1f.3: found DRSM capability at 0x1f00
> [    4.376910] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x700
> [    4.376911] sof-audio-pci 0000:00:1f.3: found SPIB capability at 0x700
> [    4.376937] sof-audio-pci 0000:00:1f.3: DMA mask is 64 bit
> [    4.376943] sof-audio-pci 0000:00:1f.3: hda global caps = 0x1009701
> [    4.376944] sof-audio-pci 0000:00:1f.3: detected 9 playback and 7 capture streams
> [    4.377044] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
> [    4.377046] sof-audio-pci 0000:00:1f.3: using HDA IRQ 133
> [    4.380271] sof-audio-pci 0000:00:1f.3: using IPC IRQ 133
> [    4.383431] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
> [    4.383435] sof-audio-pci 0000:00:1f.3: PP capability, will probe DSP later.
> [    4.383439] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000040 at  00000048
> [    4.383443] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
> [    4.388396] sof-audio-pci 0000:00:1f.3: codec_mask = 0x5
> [    4.388410] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
> [    4.388413] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000040 at  00000048
> [    4.388419] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
> [    4.388421] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
> [    4.390789] sof-audio-pci 0000:00:1f.3: HDA codec #0 probed OK: response: 10ec0892
> [    4.405091] sof-audio-pci 0000:00:1f.3: HDA codec #2 probed OK: response: 8086280b
> [    4.414325] sof-audio-pci 0000:00:1f.3: loading firmware
> [    4.417740] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000002 at  00000048
> [    4.417744] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
> [    4.417751] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
> [    4.417753] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000004 at  00000044
> [    4.417754] sof-audio-pci 0000:00:1f.3: booting DSP firmware
> [    4.417815] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
> [    4.417830] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
> [    4.417834] sof-audio-pci 0000:00:1f.3: period_bytes:0x0
> [    4.417835] sof-audio-pci 0000:00:1f.3: periods:1
> [    4.417893] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0f successful
> [    4.417907] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0e successful
> [    4.417911] sof-audio-pci 0000:00:1f.3: unstall/run core: core_mask = 1
> [    4.417917] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
> [    4.420087] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x80000000 successful
> [    4.420092] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf010e0e successful
> [    4.420131] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
> [    4.420132] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
> [    4.487482] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x5000001 successful
> [    7.487613] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x80000012 timedout
> [    7.488163] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x140000 successful
> [    7.488166] sof-audio-pci 0000:00:1f.3: error: load fw failed ret: -110
> [    7.488188] sof-audio-pci 0000:00:1f.3: unknown ROM status value 80000012
[... cut ...]
> [    7.488198] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware -110
> [    7.538981] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1c003c timedout
> [    7.538985] sof-audio-pci 0000:00:1f.3: error: reset enter failed: core_mask f adspcs 0x1c003c
> [    7.538989] sof-audio-pci 0000:00:1f.3: error: dsp core reset failed: core_mask f
> [    7.539113] sof-audio-pci 0000:00:1f.3: Turning i915 HDAC power off
> [    7.539131] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -110

plbossart, 2021-02-04:
> Note that in the first report for this issue I had a different kernel - 5.10.0-rc5+ built according to SOF documentation, whereas here it's the recommended UP kernel from repository ppa:aaeon-cm/5.4-upboard.

I don't think this can work. the minimal requirement to get SOF to work is version 5.5.9 (or 11). There is no problem using the latest kernel, I recommend you use our tree (topic/sof-dev branch) so that you don't have any issues.

worldgeek, 2021-02-04:
Thanks for the tip @plbossart, I'll use the v1.7-rc1 tag just added by @lgirdwood and try again.

Just to restate that this problem of loading community signed firmware was first noted from a topic/sof-dev kernel 5.10.0-rc5+ built according to SOF documentation.
I tried the older 5.4 kernel in case it was a regression. The Intel signed firmware does appear to work on the 5.4 kernel e.g. sof-ctl can be used to change EQ parameters with the expected change in audio output.

lgirdwood, 2021-02-04:
@plbossart btw, are you able to attach a community signed binary that works on your UP board here ? and are you using the meu tool to sign ? I am wondering here if rimage is not setting the "I am using key N" variable in the manifest.

plbossart, 2021-02-04:
@worldgeek I just tried with https://github.com/thesofproject/linux/pull/2742 and sof-bin v1.6.1 and I have no problem booting.

[dmesg_up_extreme.txt](https://github.com/thesofproject/sof-bin/files/5928986/dmesg_up_extreme.txt)

plbossart, 2021-02-04:
@lgirdwood I am not doing anything other than 

export SOF_VERSION=v1.6.1
./go.sh

if I compile myself it works as well. You know me, I can't use MEU tools :-) 
here's the binary 
[sof-cnl.ri.gz](https://github.com/thesofproject/sof-bin/files/5929019/sof-cnl.ri.gz)

worldgeek, 2021-02-05:
@plbossart @lgirdwood Community firmware is now loading.
The problem appears to be simply that my board's DMI name did not appear in the list of names for which community key is used. Presumably trying to force use of a community firmware by copying it over the default just ended up confusing the firmware loader.

This line looked interesting since it never appeared in the UP Xtreme log.
[28.207985] sof-audio-pci 0000:00:1f.3: Platform uses community key, changed fw path to intel/sof/community

I modified ./linux/sound/soc/sof/sof-pci-dev.c to add the board name for the UP Xtreme.
diff --git a/sound/soc/sof/sof-pci-dev.c b/sound/soc/sof/sof-pci-dev.c
index 7757463bd81a..e34c30307175 100644
```
--- a/sound/soc/sof/sof-pci-dev.c
+++ b/sound/soc/sof/sof-pci-dev.c
@@ -58,6 +58,13 @@ static const struct dmi_system_id sof_tplg_table[] = {
 };
 
 static const struct dmi_system_id community_key_platforms[] = {
+       {
+               .ident = "Up Xtreme",
+               .matches = {
+                       DMI_MATCH(DMI_SYS_VENDOR, "AAEON"),
+                       DMI_MATCH(DMI_BOARD_NAME, "UP-WHL01"),
+               }
+       },
        {
                .ident = "Up Squared",
                .matches = {

```

plbossart, 2021-02-05:
> @plbossart @lgirdwood Community firmware is now loading.
> The problem appears to be simply that my board's DMI name did not appear in the list of names for which community key is used. Presumably [...]

That's what thesofproject/linux#2742 does :-)

worldgeek, 2021-02-05:
> That's what thesofproject/linux#2742 does :-)

Haha! I missed that little detail. Was wondering what the DMI NAME was on @plbossart's board. :-)

marc-hb, 2021-09-28:
@worldgeek is this still an issue?

lenghonglin, 2021-12-27:
the same question, and already update kernel to 5.13.

but I am not sure whether to upgrade the BIOS

plbossart, 2022-01-10:
@lenghonglin you should upgrade the SOF firmware to 1.9 at least