thesofproject/sof-bin #36: Any plans for mircophone Noise supression?

joakim-tjernlund, 2021-03-24:


lgirdwood, 2021-03-25:
@joakim-tjernlund can you be more specific ? @singalsu fyi.

joakim-tjernlund, 2021-03-25:
Not really, I only know that on our new Lenovo T14 with SOF one can hear some noise(white?) when using the built-in mic.

plbossart, 2021-03-25:
@joakim-tjernlund please file an issue in the Linux tree and add your dmesg log with the following file added as /etc/modprobe.d/sof-dyndng.sof 
[sof-dyndbg.conf.txt](https://github.com/thesofproject/sof-bin/files/6206881/sof-dyndbg.conf.txt)

You may be experiencing known issues with the 2 mic support on some Lenovo platforms who use the PDM1 interface.

see e.g. https://github.com/thesofproject/linux/issues/2795

singalsu, 2021-03-26:
@joakim-tjernlund There's a microphone beamformer and if properly set up it will attenuate a bit the noise but it also sets a fixed direction for microphones. Out-of-beam directions get attenuation. It's feasible with notebooks if the direction of interest is towards camera look direction. Currently that needs a custom topology for specific microphones distance and microphones count. If you are willing to experiment with non-mature component I can help to set it up. There will be later in future a noise suppressor for voice.

If you get sound from microphone capture it's worth to check that every volume/gain is sane. E.g. no gains competing against each other. If can report which topology is loaded (or attach dmesg) and which device you are capturing would help.

marc-hb, 2021-07-28:
> @joakim-tjernlund please file an issue in the Linux tree

Transferring to Linux tracker