thesofproject/sof-bin #41: [FEATURE] support .xz compressed firmware

plbossart, 2021-05-06:
Fedora 34 supports compressed firmware https://fedoraproject.org/wiki/Changes/CompressKernelFirmware. This is used for existing firmware released by Intel (atom and Skylake). This capability exists since 5.3, just before the SOF driver was upstream or usable.

Should the sof-bin installer optionally compress the .ri and .tplg files with the .xz format? We start having quite a few of them.
The default Kconfig for SOF development already support this capability.

@perexg @marc-hb @lgirdwood @jwrdegoede FYI.

perexg, 2021-05-06:
The alsa-sof-firmware package does not use the sof-bin installer. I'll update the rpm spec file to compress the firmware files for 1.7 release.

marc-hb, 2021-05-07:
`git` already uses compression, so this would not make any significant git performance difference. As a sample datapoint I created two new, test git repo with the entire v1.7 release compressed with `xz` and not. The `.git/` size difference is less than 5%.

> Should the sof-bin installer ... Fedora does not use the sof-bin installer ...

The installer is now in `sof/installer/` so any developer can use it, there's no installer code in `sof-bin` needed any more at this moment. See #40. I use the `sof/installer/` to compile and deploy hundreds of times a day. "ktest-style" but much faster.

> .... optionally compress the .ri and .tplg files with the .xz format?

You mean as an addition to #40 which is not reviewed yet? Seems doable, my only concern is that we already have symbolic links and https://fedoraproject.org/wiki/Changes/CompressKernelFirmware mentions more symlinks so I'm afraid about incompatibilities or that people might get lost.

marc-hb, 2021-10-05:
> I'll update the rpm spec file to compress the firmware files for 1.7 release.

@perexg how did that go? As simple as running `xz` on each `.ri` file, done?

> Should the sof-bin installer optionally compress the .ri and .tplg files with the .xz format?

@plbossart I feel like this should be left to Linux distributions considering they own kernel configuration and considering the new sof-bin "installer" is now just a dumb recursive copy script that's not even required. Fedora does not seem to use it: https://github.com/thesofproject/sof-bin/issues/46#issuecomment-855199725

plbossart, 2022-01-18:
will be left as an exercise for the distro package maintainer(s)

closed 2022-01-18 by plbossart