thesofproject/sof-bin #51: v1.8 release notes for the signed firmware files - Alderlake

perexg, 2021-07-01:
The Alderlake firmware (.ri) files are *NOT* available in v1.8. It would be nice to mention this in the binary release notes, otherwise the users may be confused.

plbossart, 2021-07-01:
> The Alderlake firmware (.ri) files are _NOT_ available in v1.8. It would be nice to mention this in the binary release notes, otherwise the users may be confused.

Alderlake support is planned for a later release. Is there really an impact to users at this point?

perexg, 2021-07-01:
The impact is just a confusion. Distributions take the release notes and the Alderlake is mentioned there, but firmware files for this platform are missing. Also, kernel drivers refer to those files.

lgirdwood, 2021-07-01:
So support is in the FW codebase, but there is no signed binary as the signing mechanism is not available yet. I'll remove this from the notes.

perexg, 2021-07-01:
> So support is in the FW codebase, but there is no signed binary as the signing mechanism is not available yet. I'll remove this from the notes.

Yes, exactly. Also, it's a bit questionable to release the compiled topology files when the signed / public DSP binary code is not available.

marc-hb, 2021-10-06:
To keep things simple I just added a link from https://github.com/thesofproject/sof-bin/releases/tag/v1.9 to https://github.com/thesofproject/sof/releases/tag/v1.9. I think the "inconvenience" of one extra click is acceptable. I'll update the HOWTO.

Closing this old 1.8 issue. Any problem with 1.9 please use #57 or file a new issue.

closed 2021-10-06 by marc-hb