thesofproject/sof-bin #52: How to uninstall?

dragossusi, 2021-07-05:
I tried this project but doesn't work on latest Intel processors.
How do I remove this?

marc-hb, 2021-07-05:
Which processor and what failure do you have?

The "installer" merely copies files, into 3 directories:
https://github.com/thesofproject/sof-bin/blob/main/README.md
So you can delete these directories are restore your backup.

If you forgot to make a backup, you can re-install the linux-firmware package from your distribution.

dragossusi, 2021-07-06:
I didn't backup anything.

lgirdwood, 2021-07-06:
> I tried this project but doesn't work on latest Intel processors.
> How do I remove this?

Which processor ? It works on processors from Baytrail to Tigerlake (latest).

marc-hb, 2021-07-28:
@dragossusi ok to close?

dragossusi, 2021-07-29:
yes, the problem we had was with audio and microphone, they didnt work on a dell Workstation Dell Precision 5750 17" FHD+ i7-10750H

marc-hb, 2021-07-29:
> i7-10750H

Seems to be CML

closed 2021-08-05 by marc-hb