thesofproject/sof-bin #59: sof-glk.ri unsupported or invalid extended manifest: -22

edgardogho, 2021-10-04:
Hello,
I have a motherboard with Intel J4125 (AIO Computer) and uses 8086:3198 (PCI).
I'm building the latest kernel (5.14.9 and 5.15-rc3) and using the SOF-bin v1.9.x.
I can't get the firmware to load. DMESG says:
[    2.291102] igb: Intel(R) Gigabit Ethernet Network Driver
[    2.291106] igb: Copyright (c) 2007-2014 Intel Corporation.
[    2.301446] sof-audio-pci-intel-apl 0000:00:0e.0: DSP detected with PCI class/subclass/prog-if 0x040100
[    2.302629] sof-audio-pci-intel-apl 0000:00:0e.0: use msi interrupt mode
[    2.309413] sof-audio-pci-intel-apl 0000:00:0e.0: Force to use nocodec mode
[    2.309418] sof-audio-pci-intel-apl 0000:00:0e.0: Using nocodec machine driver
[    2.309573] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware info: version 1:9:0-fa857
[    2.309576] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware: ABI 3:19:0 Kernel ABI 3:18:0
[    2.309579] sof-audio-pci-intel-apl 0000:00:0e.0: error: FW ABI is more recent than kernel
[    2.309581] sof-audio-pci-intel-apl 0000:00:0e.0: error: failed to parse sof_ext_man header type 0 size 0x50
[    2.309584] sof-audio-pci-intel-apl 0000:00:0e.0: error: firmware intel/sof/sof-glk.ri contains unsupported or invalid extended manifest: -22
[    2.309586] sof-audio-pci-intel-apl 0000:00:0e.0: error: failed to load DSP firmware -22
[    2.309747] sof-audio-pci-intel-apl 0000:00:0e.0: error: sof_probe_work failed err: -22

I've tried forcing snd_intel_dspcfg.dsp_driver= to 1,2 and 3 (the error above is using 3). With 1 no audio card is present on dmesg (other than the HDMI one), and with 2 its missing some 3198-ALASKA files.

I'm not sure why it says the FW ABI is more recent than kernel.. happens on kernel 5.15-rc3.

Any pointers into how to get this chip working?
Thanks

marc-hb, 2021-10-04:
Can you run `sudo find /lib/ -name sof-glk.ri` ?

Can you try older releases?

edgardogho, 2021-10-04:
> Can you run `sudo find /lib/ -name sof-glk.ri` ?
> 
> Can you try older releases?

Sure, 
/lib/firmware/intel/sof-v1.9/community/sof-glk.ri
/lib/firmware/intel/sof-v1.9/sof-glk.ri
/lib/firmware/intel/sof-v1.9/intel-signed/sof-glk.ri
/lib/firmware/intel/sof-v1.9-rc1/community/sof-glk.ri
/lib/firmware/intel/sof-v1.9-rc1/sof-glk.ri
/lib/firmware/intel/sof-v1.9-rc1/intel-signed/sof-glk.ri
I basically copied all the v1.9.x bin files downloaded from github and ended up with  sof and sof-tplg inside the initramfs. The slinks point to sof-v1.9 and sof-tplg-v1.9 folders so its using the latest ones.

I will try with 1.8 and report back.

plbossart, 2021-10-04:
also please provide a link to the result of 'alsa-info' and clarify if this is the *first* time you try SOF?

GLK is typically only used for Chromebooks but you may be in a case where China-based designs used the ES8116 audio codec for which support upstream is missing.

plbossart, 2021-10-04:
You may also have too strict configurations for developers only, here's what we recommend for end-user kernels

https://github.com/thesofproject/kconfig/blob/master/sof-defconfig
https://github.com/thesofproject/kconfig/blob/master/mach-driver-defconfig

edgardogho, 2021-10-04:
> also please provide a link to the result of 'alsa-info' and clarify if this is the _first_ time you try SOF?
> 
> GLK is typically only used for Chromebooks but you may be in a case where China-based designs used the ES8116 audio codec for which support upstream is missing.

ok, I will send the alsa-info shortly.
This is in fact my first time using SOF. We get AIOPC from china (mostly J3160, J3710 and now J4125) and custom build kernel and deploy a tinycore based linux distro for a product we sell.
We have been using alsa and pulseaudio with no issues (compile pulseaudio, alsa-plugins, etc), but this is the first motherboard we get with SOF, so I'm trying to add support for it on our OS build.

edgardogho, 2021-10-04:
> You may also have too strict configurations for developers only, here's what we recommend for end-user kernels
> 
> https://github.com/thesofproject/kconfig/blob/master/sof-defconfig https://github.com/thesofproject/kconfig/blob/master/mach-driver-defconfig

[output.txt](https://github.com/thesofproject/sof-bin/files/7278981/output.txt)
Here is a grep of the config file I'm using to build the kernel (grep SND).
I will compare with the 2 defconfigs you suggest.
Thanks for the info.

edgardogho, 2021-10-04:
> Can you run `sudo find /lib/ -name sof-glk.ri` ?
> 
> Can you try older releases?

This is what I get when I use V1.8.x:
[    0.252109] PCI-DMA: Using software bounce buffering for IO (SWIOTLB)
[    0.252111] software IO TLB: mapped [mem 0x0000000073e8a000-0x0000000077e8a000] (64MB)
[    2.348821] sof-audio-pci-intel-apl 0000:00:0e.0: DSP detected with PCI class/subclass/prog-if 0x040100
[    2.349984] sof-audio-pci-intel-apl 0000:00:0e.0: use msi interrupt mode
[    2.361505] sof-audio-pci-intel-apl 0000:00:0e.0: Force to use nocodec mode
[    2.361512] sof-audio-pci-intel-apl 0000:00:0e.0: Using nocodec machine driver
[    2.361644] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware info: version 1:8:0-9e7a8
[    2.361647] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware: ABI 3:18:1 Kernel ABI 3:18:0
[    2.361655] sof-audio-pci-intel-apl 0000:00:0e.0: unknown sof_ext_man header type 3 size 0x30
[    2.403175] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware info: version 1:8:0-9e7a8
[    2.403183] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware: ABI 3:18:1 Kernel ABI 3:18:0
[    2.673716] sof-audio-pci-intel-apl 0000:00:0e.0: Direct firmware load for intel/sof-tplg/sof-glk-nocodec.tplg failed with error -2
[    2.673726] sof-audio-pci-intel-apl 0000:00:0e.0: error: tplg request firmware intel/sof-tplg/sof-glk-nocodec.tplg failed err: -2
[    2.673728] sof-audio-pci-intel-apl 0000:00:0e.0: you may need to download the firmware from https://github.com/thesofproject/sof-bin/
[    2.673730] sof-audio-pci-intel-apl 0000:00:0e.0: error: failed to load DSP topology -2
[    2.673732] sof-audio-pci-intel-apl 0000:00:0e.0: ASoC: error at snd_soc_component_probe on 0000:00:0e.0: -2
[    2.673742] sof-nocodec sof-nocodec: ASoC: failed to instantiate card -2
[    2.674137] sof-nocodec: probe of sof-nocodec failed with error -2

when I run a find for sof-glk* I get:
/lib/firmware/intel/sof-v1.8/sof-glk.ri
/lib/firmware/intel/sof-v1.8/sof-glk.ldc
/lib/firmware/intel/sof-v1.8/community/sof-glk.ri
/lib/firmware/intel/sof-v1.8/intel-signed/sof-glk.ri
/lib/firmware/intel/sof-tplg-v1.8/sof-glk-da7219-kwd.tplg
/lib/firmware/intel/sof-tplg-v1.8/sof-glk-rt5682.tplg
/lib/firmware/intel/sof-tplg-v1.8/sof-glk-da7219.tplg
/lib/firmware/intel/sof-tplg-v1.8/sof-glk-cs42l42.tplg
/lib/firmware/intel/sof-tplg-v1.8-rc2/sof-glk-da7219-kwd.tplg
/lib/firmware/intel/sof-tplg-v1.8-rc2/sof-glk-rt5682.tplg
/lib/firmware/intel/sof-tplg-v1.8-rc2/sof-glk-da7219.tplg
/lib/firmware/intel/sof-tplg-v1.8-rc2/sof-glk-cs42l42.tplg
/lib/firmware/intel/sof-v1.8-rc2/sof-glk.ri
/lib/firmware/intel/sof-v1.8-rc2/sof-glk.ldc
/lib/firmware/intel/sof-v1.8-rc2/community/sof-glk.ri
/lib/firmware/intel/sof-v1.8-rc2/intel-signed/sof-glk.ri

marc-hb, 2021-10-04:
> [ 2.673726] sof-audio-pci-intel-apl 0000:00:0e.0: error: tplg request firmware intel/sof-tplg/sof-glk-nocodec.tplg failed err: -2

There was indeed no `sof-glk-nocodec.tplg` in release v1.8.x, it was added by commit v1.9-rc1~373 (https://github.com/thesofproject/sof/pull/4378)

I don't know whether the v1.9 `sof-glk-nocodec.tplg` is compatible with v1.8 firmware but it's worth a try?

> when I run a find for sof-glk* I get:

I don't think it's going to solve your issues but for clarity and simplicity please avoid for now the `/lib/firmware/intel/sof` and `/lib/firmware/intel/sof-tplg` symbolic links for as long as you're experiencing these difficulties. Move the directories you don't want (like: everything with `v1.8-rc2` in the name) to some other backup location anywhere outside `/lib/firmware` and have only SOF version in the entire `/lib/firmware` tree at a time.

plbossart, 2021-10-04:
@edgardogho you have to revisit your kernel options based on the link I provided earlier:

"Force to use nocodec mode" is for developers only, a distribution should never enable this!

edgardogho, 2021-10-04:
> @edgardogho you have to revisit your kernel options based on the link I provided earlier:
> 
> "Force to use nocodec mode" is for developers only, a distribution should never enable this!

I'm doing that... since I never use SOF I just went ahead and select a lot of options to try pickup the sound chip. Thanks for the info, I will report if I get it going with the more carefully selected kernel config options.

edgardogho, 2021-10-04:
> @edgardogho you have to revisit your kernel options based on the link I provided earlier:
> 
> "Force to use nocodec mode" is for developers only, a distribution should never enable this!

OK, I revisited the kernel config options and remove the debug and nocodec modes...
I now have a clean 1.9 in /lib/firmware/intel/sof and sof-tplg. No symlinks, no rc, clean 1.9.
[sof-tplg.txt](https://github.com/thesofproject/sof-bin/files/7280213/sof-tplg.txt)
[sof.txt](https://github.com/thesofproject/sof-bin/files/7280214/sof.txt)

When it boots, now dmesg seems to be defaulting to snd_intel_dspcfg.dsp_driver=2 (meaning, if I don't set this on the cmdline, it shows the same output on dmesg as if I would have set the 2). The dmesg relevant output for that option is:
[    2.267966] snd_soc_skl 0000:00:0e.0: bound 0000:00:02.0 (ops 0xffffffff8133bd10)
[    2.591122] snd_soc_skl 0000:00:0e.0: Direct firmware load for 3198-ALASKA-AMI-2-tplg.bin failed with error -2
[    2.591131] snd_soc_skl 0000:00:0e.0: tplg fw 3198-ALASKA-AMI-2-tplg.bin load failed with -2, trying alternative tplg name skl_hda_dsp_generic-tplg.bin
[    2.591141] snd_soc_skl 0000:00:0e.0: Direct firmware load for skl_hda_dsp_generic-tplg.bin failed with error -2
[    2.591144] snd_soc_skl 0000:00:0e.0: tplg skl_hda_dsp_generic-tplg.bin failed with -2, falling back to dfw_sst.bin
[    2.591153] snd_soc_skl 0000:00:0e.0: Direct firmware load for dfw_sst.bin failed with error -2
[    2.591155] snd_soc_skl 0000:00:0e.0: Fallback tplg fw dfw_sst.bin load failed with -2
[    2.591157] snd_soc_skl 0000:00:0e.0: Failed to init topology!
[    2.591159] snd_soc_skl 0000:00:0e.0: ASoC: error at snd_soc_component_probe on 0000:00:0e.0: -2
[    4.409595] usbcore: registered new interface driver snd-usb-audio

If I force it on 3 I get:
[    2.251199] sof-audio-pci-intel-apl 0000:00:0e.0: DSP detected with PCI class/subclass/prog-if 0x040100
[    2.256038] sof-audio-pci-intel-apl 0000:00:0e.0: use msi interrupt mode
[    2.262100] sof-audio-pci-intel-apl 0000:00:0e.0: warning: No matching ASoC machine driver found
[    2.262106] sof-audio-pci-intel-apl 0000:00:0e.0: error: no matching ASoC machine driver found - aborting probe
[    2.262108] sof-audio-pci-intel-apl 0000:00:0e.0: error: failed to get machine info -19

If I force it on 1, I don't get any output (HDMI works, but the other audio chip does not even show on dmesg).

plbossart, 2021-10-04:
````
[ 2.262100] sof-audio-pci-intel-apl 0000:00:0e.0: warning: No matching ASoC machine driver found
````
is why we need the result of 'alsa-info'. My money is on an ES8316 device, similar problem we've been trying to solve recently with https://github.com/thesofproject/linux/pull/2962

edgardogho, 2021-10-04:
> ```
> [ 2.262100] sof-audio-pci-intel-apl 0000:00:0e.0: warning: No matching ASoC machine driver found
> ```
> 
> is why we need the result of 'alsa-info'. My money is on an ES8316 device, similar problem we've been trying to solve recently with [thesofproject/linux#2962](https://github.com/thesofproject/linux/pull/2962)

ok, I'm trying to get it to run successfully  since this motherboard runs busybox as shell, so most of the script fails like when running modprobe -c to pull the arguments for the SND modules... the busybox version does not support all that.
Is there anything in particular you want to see on the script output? I can run lspci, explore the /proc or /sys fs to find out details about the chip.

plbossart, 2021-10-04:
mostly the ACPI status - from the script this is

````bash
# Check for ACPI device status
if [ -d /sys/bus/acpi/devices ]; then
    for f in /sys/bus/acpi/devices/*/status; do
	ACPI_STATUS=$(cat $f 2>/dev/null);
	if [[ "$ACPI_STATUS" -ne 0 ]]; then
	    echo $f $'\t' $ACPI_STATUS >>$TEMPDIR/acpidevicestatus.tmp;
	fi
    done
fi
````

edgardogho, 2021-10-04:
Here is the output

/sys/bus/acpi/devices/ESSX8336:00/status 	 15
/sys/bus/acpi/devices/INT33A1:00/status 	 15
/sys/bus/acpi/devices/INT3453:00/status 	 15
/sys/bus/acpi/devices/INT3453:01/status 	 15
/sys/bus/acpi/devices/INT3453:02/status 	 15
/sys/bus/acpi/devices/INT3453:03/status 	 15
/sys/bus/acpi/devices/LNXPOWER:00/status 	 1
/sys/bus/acpi/devices/LNXPOWER:01/status 	 1
/sys/bus/acpi/devices/LNXPOWER:02/status 	 1
/sys/bus/acpi/devices/LNXPOWER:03/status 	 1
/sys/bus/acpi/devices/LNXPOWER:04/status 	 1
/sys/bus/acpi/devices/LNXPOWER:05/status 	 1
/sys/bus/acpi/devices/LNXPOWER:06/status 	 1
/sys/bus/acpi/devices/LNXPOWER:07/status 	 15
/sys/bus/acpi/devices/PNP0103:00/status 	 15
/sys/bus/acpi/devices/PNP0303:00/status 	 15
/sys/bus/acpi/devices/PNP0501:00/status 	 15
/sys/bus/acpi/devices/PNP0C0C:00/status 	 15
/sys/bus/acpi/devices/PNP0C0F:00/status 	 9
/sys/bus/acpi/devices/PNP0C0F:01/status 	 9
/sys/bus/acpi/devices/PNP0C0F:02/status 	 9
/sys/bus/acpi/devices/PNP0C0F:03/status 	 9
/sys/bus/acpi/devices/PNP0C0F:04/status 	 9
/sys/bus/acpi/devices/PNP0C0F:05/status 	 9
/sys/bus/acpi/devices/PNP0C0F:06/status 	 9
/sys/bus/acpi/devices/PNP0C0F:07/status 	 9
/sys/bus/acpi/devices/PNP0F03:00/status 	 15
/sys/bus/acpi/devices/device:09/status 	 15
/sys/bus/acpi/devices/device:0a/status 	 15
/sys/bus/acpi/devices/device:0d/status 	 15
/sys/bus/acpi/devices/device:30/status 	 15

edgardogho, 2021-10-04:
In case this helps, I cat all the files on /sys/bus/acpi/devices/ESSX8336:

adr=
0x00000000
------
hid=
ESSX8336
------
modalias=
acpi:ESSX8336:ESSX8336:
------
path=
\_SB_.PCI0.I2C1.ESSX
------
physical_node=
------
power/=
------
power_state=
D0
------
status=
15
------
subsystem=
------
uevent=
MODALIAS=acpi:ESSX8336:ESSX8336:
------
uid=
1
------

plbossart, 2021-10-04:
Bingo:

/sys/bus/acpi/devices/ESSX8336:00/status 15

-> see issue https://github.com/thesofproject/linux/issues/2955

that platform is not supported yet upstream. the Intel parts are on their way to upstream, but we're missing a codec driver update.

marc-hb, 2021-10-04:
Closing as duplicate of https://github.com/thesofproject/linux/issues/2955 ?

closed 2021-10-04 by plbossart