thesofproject/sof-bin #64: Signed files in v1.9 - increased file sizes by 40%

perexg, 2021-11-15:
I just noted that the signed firmware (.ri) file sizes are increased by 40% . It may show some issue (missing strip for the debug symbols or so) ?

marc-hb, 2021-11-16:
First, thanks a lot for keeping an eye on this. Much appreciated.

All the firmware files are signed. The `community` files are signed with a not secret private key. @perexg which other files have NOT increased?

I did a little bit of`./sof/tools/sof_ri_info/sof_ri_info.py`  and bisection and I think the big file size increase  is not in `.text`, it happened only in `.rodata` and `.bss`. @keyonjie can you confirm?

Example: 
 ```
./sof/tools/sof_ri_info/sof_ri_info.py -v ./sof-bin/v1.9.x/sof-v1.9/sof-cnl.ri 
  l2 hpsram                           0xbe000000 (3031040 + 114688  96.35% used)
    BASEFW.text                         0xbe00c000 (196608)
    BASEFW.rodata                       0xbe03c000 (299008)
    BASEFW.bss                          0xbe085000 (2535424)
```

In v1.8 only 35% were used.

Bisection for `sof-cnl.ri` pointed at these two pull requests
https://github.com/thesofproject/sof/pull/4445 cAVS heap map refinement
https://github.com/thesofproject/sof/pull/4735 Heap refinement Part 2 -- make full use of HPSRAM in buffer zone

perexg, 2021-11-16:
> All the firmware files are signed. The `community` files are signed with a not secret private key. @perexg which other files have NOT increased?

I apologize for the vague description. The .ri files in the `community` + `intel-signed` subdirectories are +40%. Other file sizes changed only slightly, so our package inspection tool did not report them.

keyonjie, 2021-11-16:
@marc-hb @perexg just checked and found this is caused by the increase of usage of the .shared_data section which is put into the FW binary. 

Thanks for reporting, let's transfer the issue to the sof repo and fixes needed for it there.