thesofproject/sof-bin #73: sof-jsl-es8336.tplg missing

tomty89, 2022-03-10:
Although as of linux 5.16.13 (and sof-bin 2.0), analog sound still doesn't seem to work at all on my laptop. However, DP sound seems to work fine with `sof-jsl-es8336.tplg` provided [here](https://github.com/thesofproject/linux/issues/3210#issuecomment-942790456). Is there a reason it isn't provided/included here?

Btw although it shows/works on ALSA, pulseaudio does not show card profile for its HDMI/DP card devices:

`aplay -l`:
```
**** List of PLAYBACK Hardware Devices ****
card 0: sofessx8336 [sof-essx8336], device 0: ES8336 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofessx8336 [sof-essx8336], device 5: HDMI 1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofessx8336 [sof-essx8336], device 6: HDMI 2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofessx8336 [sof-essx8336], device 7: HDMI 3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
`pacmd list-cards`
```
1 card(s) available.
    index: 0
	name: <alsa_card.pci-0000_00_1f.3-platform-sof-essx8336>
	driver: <module-alsa-card.c>
	owner module: 6
	properties:
		alsa.card = "0"
		alsa.card_name = "sof-essx8336"
		alsa.long_card_name = "ALLDOCUBE-i1405-Defaultstring-ALLDOCUBE"
		alsa.driver_name = "snd_soc_sof_es8336"
		device.bus_path = "pci-0000:00:1f.3-platform-sof-essx8336"
		sysfs.path = "/devices/pci0000:00/0000:00:1f.3/sof-essx8336/sound/card0"
		device.bus = "pci"
		device.vendor.id = "8086"
		device.vendor.name = "Intel Corporation"
		device.product.id = "4dc8"
		device.string = "0"
		device.description = "sof-essx8336"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card-pci"
	profiles:
		input:stereo-fallback: Stereo Input (priority 51, available: no)
		output:stereo-fallback: Stereo Output (priority 37868, available: unknown)
		output:stereo-fallback+input:stereo-fallback: Stereo Output + Stereo Input (priority 5151, available: unknown)
		off: Off (priority 0, available: unknown)
	active profile: <output:stereo-fallback>
	sinks:
		alsa_output.pci-0000_00_1f.3-platform-sof-essx8336.stereo-fallback/#0: sof-essx8336 Stereo
	sources:
		alsa_output.pci-0000_00_1f.3-platform-sof-essx8336.stereo-fallback.monitor/#0: Monitor of sof-essx8336 Stereo
	ports:
		analog-input-headset-mic: Headset Microphone (priority 8800, latency offset 0 usec, available: no)
			properties:
				device.icon_name = "audio-input-microphone"
		analog-output-speaker: Speakers (priority 10000, latency offset 0 usec, available: unknown)
			properties:
				device.icon_name = "audio-speakers"
		analog-output-headphones: Headphones (priority 9900, latency offset 0 usec, available: no)
			properties:
				device.icon_name = "audio-headphones"
```
Should I file an issue for that somewhere else? (I don't think pulseaudio upstream has anything to do with it.)

plbossart, 2022-03-10:
The patches for ES8336 support will be in 5.18, and you need the latest topologies that are not in sof-bin yet - they will be in the release 2.1.

If you want to experiment, try our development tree (topic/sof-dev branch) and the attached topologies 
[es8336-topologies-4.tar.gz](https://github.com/thesofproject/sof-bin/files/8225344/es8336-topologies-4.tar.gz)

marc-hb, 2022-03-10:
> However, DP sound seems to work fine with sof-jsl-es8336.tplg provided https://github.com/thesofproject/linux/issues/3210#issuecomment-942790456. Is there a reason it isn't provided/included here?

You haven't given any specific model number but I'm guessing the reason why it's not included is simply because no one submitted yet to the git repo the code and/or topology to support it. That or it hasn't been released yet.

I'm guessing your laptop vendor supports only Windows, correct?

marc-hb, 2022-03-10:
Closing based on @plbossart 's comment (that I saw right after I posted mine)

closed 2022-03-10 by marc-hb