thesofproject/sof-bin #86: Missing sof-apl-es8336-ssp1.tplg

Polyvyanyy, 2022-08-01:
Missing sof-apl-es8336-ssp1.tplg

`[    8.427955] sof-audio-pci-intel-apl 0000:00:0e.0: DSP detected with PCI class/subclass/prog-if info 0x040100
[    8.428270] sof-audio-pci-intel-apl 0000:00:0e.0: DSP detected with PCI class/subclass/prog-if 0x040100
[    8.428477] sof-audio-pci-intel-apl 0000:00:0e.0: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    8.435396] sof-audio-pci-intel-apl 0000:00:0e.0: use msi interrupt mode
[    8.473823] sof-audio-pci-intel-apl 0000:00:0e.0: NHLT_DEVICE_I2S detected, ssp_mask 0x3
[    8.473840] sof-audio-pci-intel-apl 0000:00:0e.0: hda codecs found, mask 4
[    8.477522] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware info: version 2:2:0-57864
[    8.477532] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware: ABI 3:22:1 Kernel ABI 3:19:1
[    8.477537] sof-audio-pci-intel-apl 0000:00:0e.0: warn: FW ABI is more recent than kernel
[    8.477554] sof-audio-pci-intel-apl 0000:00:0e.0: unknown sof_ext_man header type 3 size 0x30
[    8.529924] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware info: version 2:2:0-57864
[    8.529954] sof-audio-pci-intel-apl 0000:00:0e.0: Firmware: ABI 3:22:1 Kernel ABI 3:19:1
[    8.529979] sof-audio-pci-intel-apl 0000:00:0e.0: warn: FW ABI is more recent than kernel
[    8.555080] sof-essx8336 sof-essx8336: quirk mask 0x1
[    8.555092] sof-essx8336 sof-essx8336: quirk SSP1
[    8.556055] sof-audio-pci-intel-apl 0000:00:0e.0: Direct firmware load for intel/sof-tplg/sof-apl-es8336-ssp1.tplg failed with error -2
[    8.556065] sof-audio-pci-intel-apl 0000:00:0e.0: error: tplg request firmware intel/sof-tplg/sof-apl-es8336-ssp1.tplg failed err: -2
[    8.556067] sof-audio-pci-intel-apl 0000:00:0e.0: you may need to download the firmware from https://github.com/thesofproject/sof-bin/
[    8.556069] sof-audio-pci-intel-apl 0000:00:0e.0: error: failed to load DSP topology -2
[    8.556071] sof-audio-pci-intel-apl 0000:00:0e.0: ASoC: error at snd_soc_component_probe on 0000:00:0e.0: -2
[    8.556090] sof-essx8336 sof-essx8336: ASoC: failed to instantiate card -2
[    8.556164] sof-essx8336 sof-essx8336: snd_soc_register_card failed: -2
[    8.556168] sof-essx8336: probe of sof-essx8336 failed with error -2
`

marc-hb, 2022-08-01:
Some possibly related references:
- https://github.com/thesofproject/sof/issues/6062#issuecomment-1200874045
- https://thesofproject.github.io/latest/getting_started/intel_debug/suggestions.html#es8336-support
- https://github.com/thesofproject/linux/issues?q=is%3Aissue+label%3A%22codec+ES8336%22+

lgirdwood, 2022-08-01:
@mengdonglin pls reassign as you need. I'm not sure who takes care of this from your team.

mengdonglin, 2022-08-01:
https://github.com/thesofproject/sof/pull/6018 needs to be backported to https://github.com/thesofproject/sof/tree/stable-v2.2

junocomp, 2022-08-02:
I have a tablet that also requires sof-apl-es8336-ssp1.tplg

plbossart, 2022-08-02:
that's a new one, we've never seen SSP1 being used. This will require more topologies files to be generated. ETA end of the week, unless someone in @mengdonglin 's team beats me to it.