thesofproject/sof-bin #112: Tell what it is

blakemcbride, 2023-01-06:
Thanks for the great package!  I've been using it, and it works really well!

Small suggestion

Your README.md file provides a warning and some important specifics.  What I would suggest is that the beginning of that file should describe what the package does.  Having the other stuff afterward is great.

Thanks!

marc-hb, 2023-01-06:
Thanks @blakemcbride for the feedback! Please feel free to submit a pull request, editing the README.md does not require any advanced knowledge, in fact you are in a better position than most SOF developers because knowing "too much" often means assuming too much.

deb-intel, 2023-01-09:
@lgirdwood I am happy to read other ReadMe files.  @blakemcbride and all, please feel free to reach out if you find other ReadMe files in need of more intro material.

mengdonglin, 2023-07-07:
Closing as Readme files are ready and we sync each release with Debian and RedHat

closed 2023-07-07 by mengdonglin