thesofproject/sof-bin #120: Various issues installing sof-bin (on a TGL)

simontrimmer, 2023-04-20:
I'd mentioned this to Pierre and he suggested recording it publicly - I'd been trying to use the sof-bin repo to do some device validation but fell at the first hurdle - installation! Logs captured using one of the referenced Linux kernel versions in the v2.5 directory - "Linux-next 20230320 or newer"

v2.4.1 the install.sh complains about the missing topology directory:

root@UpX:~/git/sof-bin# ./install.sh v2.4.x/v2.4.1
./install.sh ERROR: v2.4.x/sof-tplg-v2.4.1 not found
root@UpX:~/git/sof-bin# mkdir v2.4.x/sof-tplg-v2.4.1
root@UpX:~/git/sof-bin# ./install.sh v2.4.x/v2.4.1
+ ln -sT sof-v2.4.1 /lib/firmware/intel/sof
+ ln -sT sof-tplg-v2.4.1 /lib/firmware/intel/sof-tplg
+ rsync -a v2.4.x/sof-tplg-v2.4.1 v2.4.x/sof-v2.4.1 /lib/firmware/intel/
+ rsync -a v2.4.x/tools-v2.4.1/ /usr/local/bin/
root@UpX:~/git/sof-bin#

Topology files aside, it fails to load the v2.4.1 firmware as the files are named something completely different

[    4.389138] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    4.389177] sof-audio-pci-intel-tgl 0000:00:1f.3: enabling device (0000 -> 0002)
[    4.389743] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    4.392532] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[    4.405006] sof-audio-pci-intel-tgl 0000:00:1f.3: codec #0 probe error, ret: -2
[    4.405940] sof-audio-pci-intel-tgl 0000:00:1f.3: codec #2 probe error, ret: -2
[    4.406798] sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 0
[    4.406812] sof-audio-pci-intel-tgl 0000:00:1f.3: no hda codecs found!
[    4.408386] sof-audio-pci-intel-tgl 0000:00:1f.3: Direct firmware load for intel/sof/community/sof-tgl.ri failed with error -2
[    4.408415] sof-audio-pci-intel-tgl 0000:00:1f.3: error: sof firmware file is missing, you might need to
[    4.408429] sof-audio-pci-intel-tgl 0000:00:1f.3:        download it from https://github.com/thesofproject/sof-bin/
[    4.408443] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load DSP firmware -2
[    4.410705] sof-audio-pci-intel-tgl 0000:00:1f.3: error: sof_probe_work failed err: -2

root@UpX:/lib/firmware/intel/sof# find /lib/firmware/intel/sof/community/
/lib/firmware/intel/sof/community/
/lib/firmware/intel/sof/community/sofz-tgl.ri
/lib/firmware/intel/sof/community/sofz-adl-s.ri
/lib/firmware/intel/sof/community/sofz-rpl-s.ri
/lib/firmware/intel/sof/community/sofz-adl.ri
/lib/firmware/intel/sof/community/sofz-rpl.ri
/lib/firmware/intel/sof/community/sofz-tgl-h.ri
root@UpX:/lib/firmware/intel/sof#

v2.5 installs, but with an additional directory component so the kernel can't find the firmware.

[   19.989822] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[   19.989863] sof-audio-pci-intel-tgl 0000:00:1f.3: enabling device (0000 -> 0002)
[   19.990920] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[   19.994115] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[   20.009144] sof-audio-pci-intel-tgl 0000:00:1f.3: codec #0 probe error, ret: -2
[   20.011111] sof-audio-pci-intel-tgl 0000:00:1f.3: codec #2 probe error, ret: -2
[   20.012174] sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 0
[   20.012193] sof-audio-pci-intel-tgl 0000:00:1f.3: no hda codecs found!
[   20.015197] sof-audio-pci-intel-tgl 0000:00:1f.3: Direct firmware load for intel/sof/community/sof-tgl.ri failed with error -2
[   20.015229] sof-audio-pci-intel-tgl 0000:00:1f.3: error: sof firmware file is missing, you might need to
[   20.015247] sof-audio-pci-intel-tgl 0000:00:1f.3:        download it from https://github.com/thesofproject/sof-bin/
[   20.015265] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load DSP firmware -2
[   20.019606] sof-audio-pci-intel-tgl 0000:00:1f.3: error: sof_probe_work failed err: -2

root@UpX:~# find /lib/firmware/intel/sof/ -name sof-tgl.ri
/lib/firmware/intel/sof/tgl/community/sof-tgl.ri
/lib/firmware/intel/sof/tgl/intel-signed/sof-tgl.ri
root@UpX:~# ls -l /lib/firmware/intel/sof/
total 16
drwxr-xr-x 3 root root 4096 Apr 17 12:14 adl
drwxr-xr-x 3 root root 4096 Apr 17 12:14 adl-n
drwxr-xr-x 3 root root 4096 Apr 17 12:14 rpl
drwxr-xr-x 4 root root 4096 Apr 17 12:14 tgl
root@UpX:~#

When I adjusted the /lib/firmware/intel/sof link to be to sof-v2.5/tgl the DSP failed to start

[    7.372732] sof-audio-pci-intel-tgl 0000:00:1f.3: hda_cl_copy_fw: timeout with rom_status_reg (0x80000) read
[    7.372820] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[    7.372866] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware download failed
[    7.372906] sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_IN_PROGRESS (2)
[    7.372956] sof-audio-pci-intel-tgl 0000:00:1f.3: 0x80000006: module: ROM, state: VERIFY_FEATURE_MASK, not running
[    7.373013] sof-audio-pci-intel-tgl 0000:00:1f.3: error code: 0x14 (unknown)
[    7.373065] sof-audio-pci-intel-tgl 0000:00:1f.3: extended rom status:  0x80000006 0x14 0x0 0x0 0x0 0x0 0x2510113 0x0
[    7.373115] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[    7.373195] sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to start DSP
[    7.373267] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to boot DSP firmware -110

So all very awkward - I'll test with some of the other referenced kernel tags but as this seems strange enough to warrant posting in advance of those results as it could be mismatched installation documentation?

marc-hb, 2023-04-20:
Thanks @simontrimmer for the feedback. The v2.4 is a special, "opt-in", preview/beta release for "experts" only. 

If you just want your audio to work please do NOT git clone the sof-bin repository but use the latest v2.2.4 tarball instead:
https://github.com/thesofproject/sof-bin/releases

Quoting the start of the `README.md` at https://github.com/thesofproject/sof-bin

> For a complete release that includes all Intel products including older ones you MUST use official release tarballs found here:

marc-hb, 2023-04-20:
If you feel adventurous (thanks!) then don't use the 2.4 release but use the newer 2.5 release instead. There are detailed instructions in the v2.5x directory. Avoid the `install.sh` script and `rsync` directories directly instead, the `install.sh` script is tested only with tarballs.

simontrimmer, 2023-04-20:
Hi @marc-hb
Yep my other systems are at v2.2 but unfortunately I need to be adventurous - I'm developing drivers for new hardware and was asked to check how it behaves with the new firmware that has IPCv4. I thought there was something a bit iffy with the install script - looking at it again it looks like the files are intended for a completely different location in the firmware repo (it's installing into intel/sof/ but it'll be searched from within intel/avs[-tplg]/) 

I'll have another look tomorrow

simontrimmer, 2023-04-21:
Yep using the different search path gets us a bit further on - I'll close this

closed 2023-04-21 by simontrimmer