thesofproject/sof-bin #121: Matebook 14 2021 Intel; Firmware missing

TigerbyteDev, 2023-04-27:
Hey, 
some Firmware for that Laptop is missing. I figured out the Output Part already. I already explained the whole thing in this Bugzilla report: https://bugzilla.redhat.com/show_bug.cgi?id=2190156

perexg, 2023-04-27:
It should be in 6.2.8 then. @TigerbyteDev : Which kernel are you using? Fedora should have 6.2.12 (checked yesterday).

perexg, 2023-04-27:
Yes, and do you need the file copy / rename for this kernel ? Just remove your new file to do a test.

perexg, 2023-04-27:
@plbossart : Looking to the mentioned patch https://lore.kernel.org/alsa-devel/20230307100733.15025-1-peter.ujfalusi@linux.intel.com/ , but there is `sof-adl-es8336` prefix. This file does not exist in sof-bin-2.2.4 for ssp0:

```
$ tar tvzf sof-bin-v2.2.4.tar.gz | grep sof-adl-es8336 | cut -d ' ' -f 7-
sof-bin-v2.2.4/sof-tplg-v2.2.4/sof-adl-es8336-ssp1-hdmi-ssp02.tplg
sof-bin-v2.2.4/sof-tplg-v2.2.4/sof-adl-es8336-ssp1.tplg
```

TigerbyteDev, 2023-04-27:
> Yes, and do you need the file copy / rename for this kernel ? Just remove your new file and reboot to do a test.

After removing `sof-adl-es83x6-dmic4ch-ssp0.tplg.xz` and rebooting it doesn't work

perexg, 2023-04-27:
Thank you for the configuration, if you symlink tgl file to new adl, does things work?

```
ln -s sof-tgl-es8336-dmic4ch-ssp0.tplg.xz sof-adl-es8336-dmic4ch-ssp0.tplg.xz
```

The new kernels should look to `sof-adl-es8336` not `sof-adl-es83x6` files.

TigerbyteDev, 2023-04-27:
Do you know how to get input working by any chance?

plbossart, 2023-04-27:
> @mengdonglin : A candidate for new binary release ?

indeed looks like a miss.

TigerbyteDev, 2023-04-27:
> > Do you know how to get input working by any chance?
> 
> if you have digital microphones indeed, you can check what the device number is with arecord -l, and then `arecord -Dhw:0,<number> -c4 -r48000 -fS32_LE -d20 test.wav` and check if you see anything recorded.
> 
> The input with the headset is more complicated and likely not working due to GPIO issues. I gave up with the ES8336 codec support, just a complete nightmare for Linux.

yup that works, the device is on number 1. It's just really quite and not getting recognized in Gnome Settings, etc

plbossart, 2023-04-27:
the UCM support for DMIC is probably missing or not working, it's been a while since i looked into this.

You may want to check if the detection works with ``` Needle "cfg-dmics:"``` in alsa-ucm-conf/ucm2/Intel/sof-essx8336/sof-essx8336.conf

TigerbyteDev, 2023-04-27:
Where do I need to put it in that config?

TigerbyteDev, 2023-04-27:
[image]
There's already this

plbossart, 2023-04-27:
right, and that's the point I am not sure if this works. IIRC you can check the 'cfg-dmics' values with 'amixer -Dhw:0 info', but @perexg would be the expert here :-)

TigerbyteDev, 2023-04-27:
[image]

TigerbyteDev, 2023-04-27:
The Output broke now randomly btw

perexg, 2023-04-27:
The dmic configs are there. What does show the command `alsaucm -c hw:sofessx8336 dump text` ?

mengdonglin, 2023-04-28:
> sof-adl-es8336-dmic4ch-ssp0

@perexg @plbossart @TigerbyteDev This is a miss in SOF v2.2 release branch https://github.com/thesofproject/sof/tree/stable-v2.2. I will check pick them and also prepare a v2.2.5 topology binary release next week.

TigerbyteDev, 2023-04-28:
> The dmic configs are there. What does show the command `alsaucm -c hw:sofessx8336 dump text` ?

Verb.HiFi {
	Comment "Play and record HiFi quality Music"
	Device.Speaker {
		Comment Speakers
		ConflictingDevices [
			Headphones
		]
		Values {
			CaptureCTL "_ucm0001.hw:sofessx8336"
			PlaybackCTL "_ucm0001.hw:sofessx8336"
			PlaybackMasterElem DAC
			PlaybackMixerElem "Headphone Mixer"
			PlaybackPCM "_ucm0001.hw:sofessx8336"
			PlaybackPriority 100
		}
	}
	Device.Headphones {
		Comment Headphones
		ConflictingDevices [
			Speaker
		]
		Values {
			CaptureCTL "_ucm0001.hw:sofessx8336"
			JackControl "Headphone Jack"
			JackHWMute Speaker
			PlaybackCTL "_ucm0001.hw:sofessx8336"
			PlaybackMasterElem DAC
			PlaybackMixerElem "Headphone Mixer"
			PlaybackPCM "_ucm0001.hw:sofessx8336"
			PlaybackPriority 300
		}
	}
	Device.Headset {
		Comment "Headset Microphone"
		Values {
			CaptureCTL "_ucm0001.hw:sofessx8336"
			CaptureMasterElem ADC
			CaptureMixerElem "ADC PGA Gain"
			CapturePCM "_ucm0001.hw:sofessx8336"
			CapturePriority 300
			JackControl "Headset Mic Jack"
			PlaybackCTL "_ucm0001.hw:sofessx8336"
		}
	}
	Device.HDMI1 {
		Comment "HDMI / DisplayPort 1 Output"
		Values {
			CaptureCTL "_ucm0001.hw:sofessx8336"
			JackControl "HDMI/DP,pcm=5 Jack"
			PlaybackCTL "_ucm0001.hw:sofessx8336"
			PlaybackPCM "_ucm0001.hw:sofessx8336,5"
			PlaybackPriority 500
		}
	}
	Device.HDMI2 {
		Comment "HDMI / DisplayPort 2 Output"
		Values {
			CaptureCTL "_ucm0001.hw:sofessx8336"
			JackControl "HDMI/DP,pcm=6 Jack"
			PlaybackCTL "_ucm0001.hw:sofessx8336"
			PlaybackPCM "_ucm0001.hw:sofessx8336,6"
			PlaybackPriority 600
		}
	}
	Device.HDMI3 {
		Comment "HDMI / DisplayPort 3 Output"
		Values {
			CaptureCTL "_ucm0001.hw:sofessx8336"
			JackControl "HDMI/DP,pcm=7 Jack"
			PlaybackCTL "_ucm0001.hw:sofessx8336"
			PlaybackPCM "_ucm0001.hw:sofessx8336,7"
			PlaybackPriority 700
		}
	}
	Device.Dmic {
		Comment "Digital Microphone"
		Values {
			CaptureCTL "_ucm0001.hw:sofessx8336"
			CaptureChannels 4
			CaptureMixerElem Dmic0
			CapturePCM "_ucm0001.hw:sofessx8336,1"
			CapturePriority 100
			CaptureSwitch "Dmic0 Capture Switch"
			CaptureVolume "Dmic0 Capture Volume"
			PlaybackCTL "_ucm0001.hw:sofessx8336"
		}
	}
}

[adambrangenberg@matebook-adam ~]$ alsaucm -c hw:sofessx8336 dump text
Verb.HiFi {
	Comment "Play and record HiFi quality Music"
	Device.Speaker {
		Comment Speakers
		ConflictingDevices [
			Headphones
		]
		Values {
			CaptureCTL "_ucm0001.hw:sofessx8336"
			PlaybackCTL "_ucm0001.hw:sofessx8336"
			PlaybackMasterElem DAC
			PlaybackMixerElem "Headphone Mixer"
			PlaybackPCM "_ucm0001.hw:sofessx8336"
			PlaybackPriority 100
		}
	}
	Device.Headphones {
		Comment Headphones
		ConflictingDevices [
			Speaker
		]
		Values {
			CaptureCTL "_ucm0001.hw:sofessx8336"
			JackControl "Headphone Jack"
			JackHWMute Speaker
			PlaybackCTL "_ucm0001.hw:sofessx8336"
			PlaybackMasterElem DAC
			PlaybackMixerElem "Headphone Mixer"
			PlaybackPCM "_ucm0001.hw:sofessx8336"
			PlaybackPriority 300
		}
	}
	Device.Headset {
		Comment "Headset Microphone"
		Values {
			CaptureCTL "_ucm0001.hw:sofessx8336"
			CaptureMasterElem ADC
			CaptureMixerElem "ADC PGA Gain"
			CapturePCM "_ucm0001.hw:sofessx8336"
			CapturePriority 300
			JackControl "Headset Mic Jack"
			PlaybackCTL "_ucm0001.hw:sofessx8336"
		}
	}
	Device.HDMI1 {
		Comment "HDMI / DisplayPort 1 Output"
		Values {
			CaptureCTL "_ucm0001.hw:sofessx8336"
			JackControl "HDMI/DP,pcm=5 Jack"
			PlaybackCTL "_ucm0001.hw:sofessx8336"
			PlaybackPCM "_ucm0001.hw:sofessx8336,5"
			PlaybackPriority 500
		}
	}
	Device.HDMI2 {
		Comment "HDMI / DisplayPort 2 Output"
		Values {
			CaptureCTL "_ucm0001.hw:sofessx8336"
			JackControl "HDMI/DP,pcm=6 Jack"
			PlaybackCTL "_ucm0001.hw:sofessx8336"
			PlaybackPCM "_ucm0001.hw:sofessx8336,6"
			PlaybackPriority 600
		}
	}
	Device.HDMI3 {
		Comment "HDMI / DisplayPort 3 Output"
		Values {
			CaptureCTL "_ucm0001.hw:sofessx8336"
			JackControl "HDMI/DP,pcm=7 Jack"
			PlaybackCTL "_ucm0001.hw:sofessx8336"
			PlaybackPCM "_ucm0001.hw:sofessx8336,7"
			PlaybackPriority 700
		}
	}
	Device.Dmic {
		Comment "Digital Microphone"
		Values {
			CaptureCTL "_ucm0001.hw:sofessx8336"
			CaptureChannels 4
			CaptureMixerElem Dmic0
			CapturePCM "_ucm0001.hw:sofessx8336,1"
			CapturePriority 100
			CaptureSwitch "Dmic0 Capture Switch"
			CaptureVolume "Dmic0 Capture Volume"
			PlaybackCTL "_ucm0001.hw:sofessx8336"
		}
	}
}

mengdonglin, 2023-05-04:
@plbossart @plbossart @TigerbyteDev Can you please review https://github.com/thesofproject/sof-bin/pull/122? This PR is adding the missing ADL topologies.

closed 2023-05-11 by kv2019i

mengdonglin, 2023-05-12:
@plbossart @perexg @TigerbyteDev https://github.com/thesofproject/sof-bin/releases/tag/v2.2.5 is released.

Jay-716, 2023-05-14:
> The Output broke now randomly btw

Same problem with seemingly same device here.
I reported the original issue [thesofproject/linux/issue4111](https://github.com/thesofproject/linux/issues/4111).
And the issue linux#4111 fixed the kernel side problem and the fix is now merged into mainline kernel. The same issue added the missing tplgs but as mentioned in this issue, some tplgs are still missing. (I noticed that and do some symlink without reporting that to linux#4111). And I found the same problem here. Sound will stop working randomly.
Now the sof-bin 2.2.5 add all the missing tplgs and the randomly broken sound problem still exists.

As for the detailed description of this problem, I know little about sound system in linux, so my description may be inaccurate.
I am using pipewire, pipewire-pulse and wireplumber.
I tried to prevent the sound server form pausing (idle) by editing the pipewire-pulse.conf. And when the problem occurs, the sound server reports some errors and the speaker will not let out any sound.
The error looks like ```pipewire-pulse[2085]: mod.protocol-pulse: client 0x55612a21db00 [libcanberra]: ERROR command:18 (PLAY_SAMPLE) tag:11 error:25 (Input/output error)```
Then I switch the profile in pavucontrol to Off, then switch back to Pro Audio or Stero Input Output, the sound will start working again.
I have tried the original ucm and the ucm provided by a developer in linux#4111, and the problem exists with both of them. And I have tried edited .config/alsoft.conf (the config of openal), and it seems not related.

PR #122 (2023-05-11): Add v2.2.5 topology updates for Intel ADL platform with ES8336 codec
To fix #121 
Create directory sof-tplg-v2.2.5 and add these new topology files: sof-tplg-v2.2.5/
├── sof-adl-es8336-dmic2ch-ssp0.tplg
├── sof-adl-es8336-dmic2ch-ssp1.tplg
├── sof-adl-es8336-dmic2ch-ssp2.tplg
├── sof-adl-es8336-dmic4ch-ssp0.tplg
├── sof-adl-es8336-dmic4ch-ssp1.tplg
├── sof-adl-es8336-dmic4ch-ssp2.tplg
├── sof-adl-es8336-ssp0.tplg
├── sof-adl-es8336-ssp2.tplg
└── sof-adl-es8336.tplg

Add empty file v2.2.x/v2.2.5 for shell completion convenience.

The SOF PR is https://github.com/thesofproject/sof/pull/7536