thesofproject/sof-bin #126: Where is 2.6 release ?

GreyXor, 2023-07-01:
Hello,
when the 2.6 version will be released in this repo ? the latest one is the 2.2.*

marc-hb, 2023-07-01:
Same question and same answer as in #115

GreyXor, 2023-07-01:
Thanks, what will be the next opt-out version ?

lgirdwood, 2023-07-04:
@GreyXor I can speak only for Intel, but v2.7 will be the first productized release for Meteorlake (i.e. not an opt-in). We may also include Tigerlake, Alderlake and Raptorlake as non-opt in release with v2.7 (schedule permitting) otherwise they are good with existing v2.2.

GreyXor, 2023-07-04:
@lgirdwood Thank you for answering, my concern is more about AMD Rembrandt and Phoenix. https://www.phoronix.com/news/Sound-Open-Firmware-2.3

lgirdwood, 2023-07-04:
> @lgirdwood Thank you for answering, my concern is more about AMD Rembrandt and Phoenix. https://www.phoronix.com/news/Sound-Open-Firmware-2.3

Probably best for @thesofproject/amd to answer AMD releated release questions.

marc-hb, 2023-07-04:
I could be wrong but I suspect AMD's work in SOF has so far been for Chromebooks which don't rely on sof-bin releases. ChromeOS has its own validation and release branches instead (not just for SOF) 
https://chromium.googlesource.com/chromiumos/third_party/sound-open-firmware/

You also need to contact the vendor of your PC. Each PC model has  different audio configuration and requires a specific topology and a corresponding validation effort. If a PC model is only tested with Windows then there's a good chance Linux audio will not work, example:
https://github.com/thesofproject/linux/labels/codec%20ES8336

bhiregoudar, 2023-07-05:
As mentioned above, currently AMD is not releasing any pre-built binaries to end user's use directly. Currently SOF support is available for I2S end points and support is provided through the OEM's only. Please check with your vendor/OEM for compatibility and availability of SOF support.

marc-hb, 2023-07-05:
Thanks @bhiregoudar for the clarification!

closed 2023-07-05 by marc-hb