thesofproject/sof-bin #127: topology files for RPLP CRB

jianhuilee, 2023-07-04:
Hi,
May I know when the release will include the topology files in this commit? https://github.com/thesofproject/sof/commit/5fa9cb6be4cb0648293fee7fcc2939fec83668d6
Or is there anything I could do to move this forward? Thanks.

lgirdwood, 2023-07-04:
> Hi, May I know when the release will include the topology files in this commit? [thesofproject/sof@5fa9cb6](https://github.com/thesofproject/sof/commit/5fa9cb6be4cb0648293fee7fcc2939fec83668d6) Or is there anything I could do to move this forward? Thanks.

They should be in v2.6 release, I would also recommend you backport the patch to v2.2 stable branch for use on stable RPL branch.

anujm1, 2023-07-04:
> They should be in v2.6 release, I would also recommend you backport the patch to v2.2 stable branch for use on stable RPL branch.

Looks like it has been backported already: https://github.com/thesofproject/sof/commit/f2b74168146dbfcc6ce601ad5ed46b57ba20428e . Can I know when will be the next release for that branch?

mengdonglin, 2023-07-06:
@jianhuilee The release with topology for RPLP CRB is out https://github.com/thesofproject/sof-bin/releases/tag/v2.2.6
Please check :)

closed 2023-07-06 by mengdonglin