thesofproject/sof-bin #130: Why not package the dsp-enhancement topologies ?

smunaut, 2023-07-15:
There are some interesting topologies in tools/topology/topology1/dsp_enhancement/ that can be used by users but are not included in the release.

marc-hb, 2023-07-16:
Thanks @smunaut for your feedback. sof-bin is only for production files that are redistributed by Linux distributions. You are supposed to build more experimental stuff from source. This makes sure there's no confusion between the two and that it's not too easy for the vast majority of users to use the wrong file.

cc: @plbossart

Building from source should be as simple as

```shell
git clone sof
cd sof
./scripts/docker-run.sh ./scripts/build-tools.sh
./scripts/docker-run.sh ./scripts/build-tools.sh -h
```
Search https://thesofproject.github.io/latest/index.html for more

closed 2023-07-16 by marc-hb

kv2019i, 2023-09-13:
FYI to @singalsu and @lgirdwood

lgirdwood, 2023-09-14:
Update: @smunaut best to think of this as a staging area, @singalsu is working to bring all modules into default base image in v2.8 with packaged topologies to use them on newer Intel platforms.