thesofproject/sof-bin #134: Move to YEAR.MONTH[.MINOR] based versioning for sof-bin

kv2019i, 2023-09-13:
Problem statement
----------------------

Since SOF v2.2, sof-bin packaging has supported inclusion of firmware and topology files from multiple source revisions of SOF. 
The motivation for this is to allow a single sof-bin release to all necessary firmware and topology files, for all targets supported by upstream SOF binary releases.

Note: the upstream sof-bin releases do not cover all vendors -- binaries are also distributed via OEM and other channels. 

What is changing for SOF v2.7
------------------------------------

To clarify the relation of sof-bin release to a SOF firmware versions, sof-bin releases will move to a YEAR.MONTH[.MINOR] versioning.

The first new-style release will be **sof-bin-2023.09**. Current submitted binaries will include Intel binaries based on SOF2.2 (for Intel Raptor Lake and older platforms) and the soon to be released SOF2.7 (for newer Intel platforms). The release is open to other vendors to submit their binaries for upstream distribution.

Package content is not going to change and distribution packages can copy the release contents verbatim to /lib/firmware (as has been done in previous sof-bin packages).

Timeline 
----------

Synrhonized with SOF2.7 released timeline https://github.com/thesofproject/sof/issues/8017

In future, sof-bin releases will be done in sync with the SOF releases (current cadence is a quarterly release).

kv2019i, 2023-09-13:
FYI @thesofproject/maintainers

perexg, 2023-09-13:
> To clarify the relation of sof-bin release to a SOF firmware versions,  sof-bin releases will move to a YEAR.MONTH[.MINOR] versioning.

You are proposing opposite. You just clarify the release date in the package name. But I see that this follows the linux firmware releases.

Fedora does not have version in the file tree (because the information was duplicate - package name + directory name) and multiple symlink levels is ugly for the standard distribution). 

Source example:

```
sof-bin-v2.2.6/sof-tplg-v2.2.6/sof-adl-rt711-l0-rt1316-l3-2ch.tplg
```

Fedora  alsa-sof-firmware-2.2.5-1.fc38.noarch: 

```
/usr/lib/firmware/intel/sof-tplg/sof-adl-rt711-l0-rt1316-l3-2ch.tplg.xz
```

I will probably create a file in /usr/share/doc/alsa-sof-firmware/ tree with the file -> SOF version mapping.

kv2019i, 2023-09-13:
Ack @perexg , the names will not be overlapping so install without versions and symlinks is possible.

We plan to use same scripting in sof-bin git to make the tarball, so that e.g. we include a manifest file into the package that specifies versions for all binary files (see https://github.com/thesofproject/sof-bin/releases/download/v2.2.6/sof-bin-v2.2.6.tar.gz and the manifest.txt it has).

marc-hb, 2023-09-13:
> You are proposing opposite.

I tried to negate the sentence you quoted but I couldn't find how :-)

> You just clarify the release date in the package name.

Adding the date is not the important change. The important change is _removing_ the misleading "v2.6" when the package content is actually a mix of v2.6 and v2.2.2 content (depending on the hardware generation).

> sof-bin-v2.2.6/sof-tplg-v2.2.6/sof-adl-rt711-l0-rt1316-l3-2ch.tplg

This is a mistake sorry. It should have always been:

> sof-bin-v2.2.6/sof-tplg/sof-adl-rt711-l0-rt1316-l3-2ch.tplg

And in the future:

> sof-bin-200X.Y/sof-tplg/sof-adl-rt711-l0-rt1316-l3-2ch.tplg

perexg, 2023-09-13:
The manifest.txt looks good. I'll include it to the doc directory in next release.

> And in the future:
>>    sof-bin-200X.Y/sof-tplg/sof-adl-rt711-l0-rt1316-l3-2ch.tplg

Yep, it seems less work for Fedora package script.

marc-hb, 2023-09-13:
> This was a design mistake in ./tarball_multi_releases.bash, sorry my bad. In the .tgz file it should have always been:

> sof-bin-v2.2.6/sof-tplg/sof-adl-rt711-l0-rt1316-l3-2ch.tplg

Fix submitted in #136, please review. Even if you don't have time to review the code, please take a quick look at how the change in the sample test tree.

kv2019i, 2023-09-15:
Draft release tarball for sof-bin-v2023.09-rc1 is now available at: 
https://github.com/thesofproject/sof-bin/releases/download/untagged-66c71a0b01ddb08cff0b/sof-bin-v2023.09-rc1.tar.gz

kv2019i, 2023-09-15:
And pre-release tagged and published at https://github.com/thesofproject/sof-bin/releases/tag/v2023.09-rc1 

Feedback is welcome. We'll follow same structure for the final v2023.09 release.

heftig, 2023-09-15:
Tried the pre-release tarball. LGTM, thanks!

mrhpearson, 2023-09-29:
Probably the wrong place to flag this, and maybe I'm missing something obvious, but were the MTL binaries missed from the new release? I'm not seeing mtl.ri in there.

heftig, 2023-09-29:
They're present:

```
/usr/lib/firmware/intel/sof-ace-tplg/sof-mtl-max98357a-rt5682-ssp2-ssp0-2ch-pdm1.tplg
/usr/lib/firmware/intel/sof-ace-tplg/sof-mtl-max98357a-rt5682-ssp2-ssp0.tplg
/usr/lib/firmware/intel/sof-ace-tplg/sof-mtl-max98357a-rt5682.tplg
/usr/lib/firmware/intel/sof-ace-tplg/sof-mtl-rt1019-rt5682.tplg
/usr/lib/firmware/intel/sof-ace-tplg/sof-mtl-rt1318-l12-rt714-l0.tplg
/usr/lib/firmware/intel/sof-ace-tplg/sof-mtl-rt711-4ch.tplg
/usr/lib/firmware/intel/sof-ace-tplg/sof-mtl-rt711-l0-rt1316-l23-rt714-l1.tplg
/usr/lib/firmware/intel/sof-ace-tplg/sof-mtl-rt712-l0-rt1712-l3.tplg
/usr/lib/firmware/intel/sof-ace-tplg/sof-mtl-sdw-cs42l42-l0-max98363-l2.tplg
/usr/lib/firmware/intel/sof-ipc4/mtl/
/usr/lib/firmware/intel/sof-ipc4/mtl/community/
/usr/lib/firmware/intel/sof-ipc4/mtl/community/sof-mtl.ri
/usr/lib/firmware/intel/sof-ipc4/mtl/intel-signed/
/usr/lib/firmware/intel/sof-ipc4/mtl/intel-signed/sof-mtl.ri
/usr/lib/firmware/intel/sof-ipc4/mtl/sof-mtl.ri
```

mrhpearson, 2023-09-29:
Oh...got it. They moved under ipc4. My bad...sorry! (scuttling back under rock)

marc-hb, 2023-09-29:
>  They moved under ipc4.

@kv2019i eventually got tired of me complaining and very recently documented all this here:

https://thesofproject.github.io/latest/getting_started/intel_debug/introduction.html#firmware-binary

https://github.com/thesofproject/sof-docs/pull/475/files

```
   * - Raptor Lake and older
     - /lib/firmware/intel/sof/sof-PLAT.ri
     - PLAT = glk, cml, ..., rpl
   * - Raptor Lake and older (community signed)
     - /lib/firmware/intel/sof/community/sof-PLAT.ri
     - PLAT = glk, cml, ..., rpl
   * - Meteor Lake and newer
     - /lib/firmware/intel/sof-ipc4/PLAT/sof-PLAT.ri
     - PLAT = mtl, lnl, ...
   * - Meteor Lake and newer (community signed)
     - /lib/firmware/intel/sof-ipc4/PLAT/community/sof-PLAT.ri
     - PLAT = mtl, lnl, ...
   * - Meteor Lake and newer Loadable Module
     - /lib/firmware/intel/sof-ipc4-lib/PLAT/UUID.bin
     - PLAT as above, UUID = UUID of the module
   * - Meteor Lake and newer Loadable Module (community signed)
     - /lib/firmware/intel/sof-ipc4-lib/PLAT/community/UUID.bin
     - PLAT as above, UUID = UUID of the module
```

marc-hb, 2023-09-30:
New test code submitted in #146 

(one more coming after this one)

marc-hb, 2023-10-03:
Second and last `tests` PR and then this can be closed:
- #147

lgirdwood, 2023-10-04:
All merged.

mrhpearson, 2023-10-04:
Just as an info note - I tweaked my Debian package stuff to work with the new format and it's all looking good. Testing it on a few platforms now, so far no issues.
I'll see if there is anything else needed once it's been reviewed by an expert, but fingers crossed will have this released there soon.

marc-hb, 2023-10-05:
> Testing it on a few platforms now, so far no issues.

This sort of feedback (good _or_ bad!) is extremely useful to us, please keep it coming.