thesofproject/sof-bin #163: Availability of AMD ACP v6.3 firmware blob

tobiasjakobi, 2024-03-18:
Hello,

I'm currently trying to get SoF working on a AMD based system, which has a ACP v6.3. I checked the sources of the Linux kernel, and [sound/soc/amd/acp-config.c](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/sound/soc/amd/acp-config.c?h=linux-6.7.y) mentions about bunch of firmware blobs for the ACP.

* `sof-acp_6_3.ri`
* `sof-acp_6_3.tplg`

I first check the SoF firmware package that my distro provides, but no sign of these files. Then I checked out sof-bin directly and searched the repo. Again nothing found.

I had a quick look at the SoF source repo, and found ACP v6.3 mentioned [here](https://github.com/thesofproject/sof/blob/main/scripts/xtensa-build-all.sh). Can someone confirm that the only way to get these blobs is to compile them from source? I did a naive attempt, but it appears to require some very special compiler toolchain.

With best wishes,
Tobias

marc-hb, 2024-03-18:
Very good questions.

You're right, so far sof-bin has been Intel-only. It could probably help release more than Intel but it doesn't right now. Even if the release process/scripts/infrastructure/etc. are ever re-used outside Intel, the release cycles and final tarballs should probably stay distinct though because they should probably align with the hardware release cycles. My 2 cents.

> I had a quick look at the SoF source repo, and found ACP v6.3 mentioned [here]

Correct: the sof source repo is definitely NOT just for Intel. For instance, the same SOF branches are used by Google for both Intel and non-Intel Chromebooks.

BTW: https://thesofproject.github.io/latest/getting_started/intel_debug/introduction.html#chromebooks-and-sof

> but it appears to require some very special compiler toolchain.

It's required for good performance but for light-weight workloads there are open-source alternatives: https://thesofproject.github.io/latest/introduction/index.html#toolchain-faq

For Zephyr (not for AMD now) you can use the Zephyr SDK
https://docs.zephyrproject.org/latest/develop/toolchains/zephyr_sdk.html

cc: @thesofproject/amd , @lgirdwood , @kv2019i , @andyross , @cujomalainey , @dbaluta

marc-hb, 2024-03-18:
BTW there is a reason why some hardware devices are meant to be used with Windows, others are meant to be used with Linux, etc. "Free software" does not imply "just works out of the box for free".

In other words: anyone who wants a device that works well with Linux must buy one where using Linux is officially supported.

The most (in)famous counter-example is probably: https://github.com/thesofproject/linux/labels/codec%20ES8336

It's different for other people who want to have fun hacking and experimenting :-)

cujomalainey, 2024-03-18:
> Hello,
>
> I'm currently trying to get SoF working on a AMD based system, which has a ACP v6.3. I checked the sources of the Linux kernel, and [...]

@bhiregou @bhiregoudar any chance AMD has plans to add their builds to the tar bar for distros?

tobiasjakobi, 2024-03-18:
Thanks Marc for the explanation. I just found it strange, in particular because the 2.8.0 release points out the newly introduced ACP v6.3 support, but no distro seems to ship with the required blobs.

I didn't do an exhaustive search, but at least Gentoo does not. And also the Ubuntu package only contains a `sof-acp.tplg`.

It's not really a huge problem, since there is also the non-SoF driver in the kernel (`CONFIG_SND_SOC_AMD_PS`).

I'll try to build this Xtensa toolchain if I find some free time. Since it's using crosstool-ng it shouldn't be too hard.

cujomalainey, 2024-03-18:
> I had a quick look at the SoF source repo, and found ACP v6.3 mentioned [here](https://github.com/thesofproject/sof/blob/main/scripts/xtensa-build-all.sh). Can someone confirm that the only way to [...]

There is a docker container you can use to build with a flavour of gcc, you can't use the full SIMD instructions but it will get you a functioning firmware that you can work with. Makes building much easier if you don't have access to the proprietary tools.

marc-hb, 2024-03-18:
> There is a docker container you can use to build with a flavour of gcc

Thanks @cujomalainey I forgot to mention that. @tobiasjakobi take a look at [8a7a1ad76af2deaa37](https://github.com/thesofproject/sof/commit/8a7a1ad76af2deaa37) for a "demo".

> I didn't do an exhaustive search, but at least Gentoo does not. And also the Ubuntu package only contains a sof-acp.tplg

AFAIK Linux distributions merely package and re-distribute https://github.com/thesofproject/sof-bin/releases (as they should?)

>  I just found it strange, in particular because the 2.8.0 release points out the newly introduced ACP v6.3 support, but no distro seems to ship with the required blobs.

If you read https://www.sofproject.org/blog/2023/12/18/sof-2-8-0-is-here/ until the very end you see "Signed binaries for selected Intel platforms are [here](https://github.com/thesofproject/sof-bin) (sof-bin) " which hints at this distinction:
- **Source**  sof release 2.8 with ACP and many other things.
- **Binary** sof-bin release v2023.12 for Intel ONLY.

Note how sof-bin releases recently changed to a date-based numbering to help. Maybe this source/binary distinction could be even clearer next time?

cc: @deb-intel 

> And also the Ubuntu package only contains a sof-acp.tplg.

@plbossart should we move all AMD topologies to the "development" folder to exclude them from sof-bin? I mean as long as sof-bin is Intel only.

bhiregoudar, 2024-03-19:
> @bhiregou @bhiregoudar any chance AMD has plans to add their builds to the tar bar for distros?
No, currently AMD don't have any plan to add builds.

bhiregoudar, 2024-03-19:
> Thanks Marc for the explanation. I just found it strange, in particular because the 2.8.0 release points out the newly introduced ACP v6.3 support, but no distro seems to ship with the required [...]

SOF support is only available on few Chrome platforms with I2S end points. So, ACP 6.3 support is not available for Chrome platforms.

closed 2024-03-19 by marc-hb

tobiasjakobi, 2024-03-19:
> SOF support is only available on few Chrome platforms with I2S end points. So, ACP 6.3 support is not available for Chrome platforms.

You mean it's **NOT** available for **NON-Chrome** platforms?

bhiregoudar, 2024-03-19:
> You mean it's **NOT** available for **NON-Chrome** platforms?

yes, not available for Non-Chrome platforms and not available on non-I2S end points