thesofproject/sof-bin #166: sof-mtl-rt711.tplg missing

kochka, 2024-04-03:
Environment
-----------
- Arch Linux (kernel 6.8.2-arch2-1)
- extra/sof-firmware 2024.03-1 ([v2024.03](https://github.com/thesofproject/sof-bin/releases/tag/v2024.03))
-  Dell XPS 9440 

Logs and console outputs
------------------------

```
[   16.260290] sof-audio-pci-intel-mtl 0000:00:1f.3: Supported default profiles
[   16.260293] sof-audio-pci-intel-mtl 0000:00:1f.3: - ipc type 1 (Requested):
[   16.260297] sof-audio-pci-intel-mtl 0000:00:1f.3:  Firmware file: intel/sof-ipc4/mtl/sof-mtl.ri
[   16.260301] sof-audio-pci-intel-mtl 0000:00:1f.3:  Topology file: intel/sof-ace-tplg/sof-mtl-rt711.tplg
[   16.260304] sof-audio-pci-intel-mtl 0000:00:1f.3: Check if you have 'sof-firmware' package installed.
[   16.260307] sof-audio-pci-intel-mtl 0000:00:1f.3: Optionally it can be manually downloaded from:
[   16.260309] sof-audio-pci-intel-mtl 0000:00:1f.3:    https://github.com/thesofproject/sof-bin/
```

I've tried to symlink sof-mtl-rt711-4ch.tplg with this unworking result :

```
ln -s /usr/lib/firmware/intel/sof-ipc4-tplg/sof-mtl-rt711-4ch.tplg /usr/lib/firmware/intel/sof-ipc4-tplg/sof-mtl-rt711.tplg
```

```
   17.657268] sof-audio-pci-intel-mtl 0000:00:1f.3: Firmware paths/files for ipc type 1:
[   17.657274] sof-audio-pci-intel-mtl 0000:00:1f.3:  Firmware file:     intel/sof-ipc4/mtl/sof-mtl.ri
[   17.657277] sof-audio-pci-intel-mtl 0000:00:1f.3:  Firmware lib path: intel/sof-ipc4-lib/mtl
[   17.657279] sof-audio-pci-intel-mtl 0000:00:1f.3:  Topology file:     intel/sof-ace-tplg/sof-mtl-rt711.tplg
[   17.658534] sof-audio-pci-intel-mtl 0000:00:1f.3: Loaded firmware library: ADSPFW, version: 2.9.0.1
[   17.676371] cs35l56 sdw:0:2:01fa:3556:01:2: supply VDD_P not found, using dummy regulator
[   17.676402] cs35l56 sdw:0:2:01fa:3556:01:2: supply VDD_IO not found, using dummy regulator
[   17.676408] cs35l56 sdw:0:2:01fa:3556:01:2: supply VDD_A not found, using dummy regulator
[   17.676667] cs35l56 sdw:0:2:01fa:3556:01:2: Got spk-id from AF01
[   17.676913] cs35l56 sdw:0:2:01fa:3556:01:3: supply VDD_P not found, using dummy regulator
[   17.676947] cs35l56 sdw:0:2:01fa:3556:01:3: supply VDD_IO not found, using dummy regulator
[   17.676954] cs35l56 sdw:0:2:01fa:3556:01:3: supply VDD_A not found, using dummy regulator
[   17.677101] cs35l56 sdw:0:2:01fa:3556:01:3: Got spk-id from AF01
[   17.677199] cs35l56 sdw:0:3:01fa:3556:01:0: supply VDD_P not found, using dummy regulator
[   17.677229] cs35l56 sdw:0:3:01fa:3556:01:0: supply VDD_IO not found, using dummy regulator
[   17.677239] cs35l56 sdw:0:3:01fa:3556:01:0: supply VDD_A not found, using dummy regulator
[   17.677344] cs35l56 sdw:0:3:01fa:3556:01:0: Got spk-id from AF01
[   17.677733] cs35l56 sdw:0:3:01fa:3556:01:1: supply VDD_P not found, using dummy regulator
[   17.677778] cs35l56 sdw:0:3:01fa:3556:01:1: supply VDD_IO not found, using dummy regulator
[   17.677783] cs35l56 sdw:0:3:01fa:3556:01:1: supply VDD_A not found, using dummy regulator
[   17.677862] cs35l56 sdw:0:3:01fa:3556:01:1: Got spk-id from AF01
[   17.766776] sof-audio-pci-intel-mtl 0000:00:1f.3: Booted firmware version: 2.9.0.1
[   17.773582] cs42l43 sdw:0:0:01fa:4243:01: devid: 0x042a43, rev: 0xa1, otp: 0x02
[   17.778917] cs35l56 sdw:0:2:01fa:3556:01:3: Cirrus Logic CS35L56 Rev B0 OTP3 fw:3.4.4 (patched=0)
[   17.778943] cs35l56 sdw:0:3:01fa:3556:01:1: Cirrus Logic CS35L56 Rev B0 OTP3 fw:3.4.4 (patched=0)
[   17.794573] sof-audio-pci-intel-mtl 0000:00:1f.3: Topology: ABI 3:29:0 Kernel ABI 3:23:0
[   17.794637] sof-audio-pci-intel-mtl 0000:00:1f.3: error: can't connect DAI dai-copier.DMIC.dmic01.capture stream dmic01
[   17.794638] sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed to add widget id 0 type 28 name : dai-copier.DMIC.dmic01.capture stream dmic01
[   17.794641] sof_sdw sof_sdw: ASoC: failed to load widget dai-copier.DMIC.dmic01.capture
[   17.794642] sof_sdw sof_sdw: ASoC: topology: could not load header: -22
[   17.794647] sof-audio-pci-intel-mtl 0000:00:1f.3: error: tplg component load failed -22
[   17.794649] sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed to load DSP topology -22
[   17.794650] sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[   17.794656] sof_sdw sof_sdw: ASoC: failed to instantiate card -22
[   17.794691] sof_sdw sof_sdw: error -EINVAL: snd_soc_register_card failed -22
[   17.794704] sof_sdw: probe of sof_sdw failed with error -22
[   17.795355] cs35l56 sdw:0:2:01fa:3556:01:2: Cirrus Logic CS35L56 Rev B0 OTP3 fw:3.4.4 (patched=0)
[   17.795362] cs35l56 sdw:0:3:01fa:3556:01:0: Cirrus Logic CS35L56 Rev B0 OTP3 fw:3.4.4 (patched=0)
[   17.799055] cs35l56 sdw:0:3:01fa:3556:01:1: Slave 1 state check1: UNATTACHED, status was 1
[   17.799058] cs35l56 sdw:0:3:01fa:3556:01:0: Slave 2 state check1: UNATTACHED, status was 1
[   17.799086] cs35l56 sdw:0:2:01fa:3556:01:3: Slave 1 state check1: UNATTACHED, status was 1
[   17.799087] cs35l56 sdw:0:2:01fa:3556:01:2: Slave 2 state check1: UNATTACHED, status was 1
[   17.803282] cs42l43 sdw:0:0:01fa:4243:01: Slave 6 state check1: UNATTACHED, status was 1
[   17.836826] Adding alias for supply vdd-amp,(null) -> vdd-amp,sdw:0:0:01fa:4243:01
[   19.348471] ucsi_acpi USBC000:00: unknown error 0
[   19.348479] ucsi_acpi USBC000:00: UCSI_GET_PDOS failed (-5)
```

plbossart, 2024-04-03:
This looks like the same issue as https://github.com/thesofproject/linux/issues/4879, where a spurious ACPI device causes the wrong topology to be selected.

It's really not that the topology is missing, more that the kernel you have selects the wrong topology. You need to apply the patch "ASoC: Intel: mtl-match: Add cs42l43_l0 cs35l56_l23 for MTL"

kochka, 2024-04-03:
Yes I've just found this issue at the same time you answered ^^ Thx, I'll check that out

closed 2024-04-03 by kochka