thesofproject/sof-bin #170: Security - Defaults

MuldersMakes, 2024-07-15:
default install of coming with distribution debian bookworm, have no owner or group.

root@**# find / -xdev \( -nouser -o -nogroup \) -print
/usr/lib/firmware/intel
/usr/lib/firmware/intel/sof-tplg
/usr/lib/firmware/intel/sof-tplg/sof-adl-nocodec.tplg
/usr/lib/firmware/intel/sof-tplg/sof-rpl-rt711.tplg
/usr/lib/firmware/intel/sof-tplg/sof-icl-rt711-rt1308-rt715.tplg
/usr/lib/firmware/intel/sof-tplg/sof-apl-keyword-detect.tplg
/usr/lib/firmware/intel/sof-tplg/sof-tgl-rt715-rt711-rt1308-mono.tplg
/usr/lib/firmware/intel/sof-tplg/sof-jsl-es8336-dmic2ch-ssp0.tplg
/usr/lib/firmware/intel/sof-tplg/sof-byt-rt5682-ssp0.tplg
/usr/lib/firmware/intel/sof-tplg/sof-cht-rt5682.tplg
/usr/lib/firmware/intel/sof-tplg/sof-icl-rt700-4ch.tplg
/usr/lib/firmware/intel/sof-tplg/sof-icl-es8336-dmic4ch-ssp1.tplg
/usr/lib/firmware/intel/sof-tplg/sof-apl-es8336-dmic4ch-ssp5.tplg
/usr/lib/firmware/intel/sof-tplg/sof-glk-es8336-dmic4ch-ssp1.tplg
/usr/lib/firmware/intel/sof-tplg/sof-mt8195-mt6359-max98390-rt5682-rtnr.tplg
/usr/lib/firmware/intel/sof-tplg/sof-imx8-compr-wm8960-mixer.tplg
/usr/lib/firmware/intel/sof-tplg/sof-apl-es8336-ssp5.tplg
/usr/lib/firmware/intel/sof-tplg/sof-jsl-cs42l42-mx98360a.tplg
/usr/lib/firmware/intel/sof-tplg/sof-rpl-rt711-4ch.tplg
/usr/lib/firmware/intel/sof-tplg/sof-whl-rt5682.tplg
/usr/lib/firmware/intel/sof-tplg/sof-tgl-es8336-dmic2ch-ssp0.tplg
/usr/lib/firmware/intel/sof-tplg/sof-adl-max98360a-rt5682-rtnr.tplg
/usr/lib/firmware/intel/sof-tplg/sof-mt8195-mt6359-max98390-rt5682.tplg
/usr/lib/firmware/intel/sof-tplg/sof-adl-rt711-l0-rt1308-l12-rt715-l3.tplg
/usr/lib/firmware/intel/sof-tplg/sof-adl-max98390-rt5682-rtnr.tplg
/usr/lib/firmware/intel/sof-tplg/sof-adl-max98390-rt5682.tplg
/usr/lib/firmware/intel/sof-tplg/sof-rpl-rt711-l0-rt1318-l12.tplg
/usr/lib/firmware/intel/sof-tplg/sof-adl-es8336-ssp1-hdmi-ssp02.tplg
/usr/lib/firmware/intel/sof-tplg/sof-cht-rt5670.tplg
/usr/lib/firmware/intel/sof-tplg/sof-adl-es8336-ssp1.tplg
/usr/lib/firmware/intel/sof-tplg/sof-byt-rt5651-ssp0.tplg
/usr/lib/firmware/intel/sof-tplg/sof-cml-rt711-rt1308-mono-rt715.tplg
/usr/lib/firmware/intel/sof-tplg/sof-tgl-max98357a-rt5682-pdm1-google-aec.tplg
/usr/lib/firmware/intel/sof-tplg/sof-rpl-rt711-l2-rt1316-l01.tplg
/usr/lib/firmware/intel/sof-tplg/sof-rpl-nocodec.tplg
/usr/lib/firmware/intel/sof-tplg/sof-tgl-rt711-l0-rt1316-l1-mono-rt714-l3.tplg
/usr/lib/firmware/intel/sof-tplg/sof-apl-pcm512x.tplg
/usr/lib/firmware/intel/sof-tplg/sof-apl-es8336-dmic4ch-ssp2.tplg
/usr/lib/firmware/intel/sof-tplg/sof-tgl-sdw-max98373-rt5682.tplg
/usr/lib/firmware/intel/sof-tplg/sof-mt8195-mt6359-rt1019-rt5682.tplg
/usr/lib/firmware/intel/sof-tplg/sof-tgl-max98373-rt5682-igonr.tplg
/usr/lib/firmware/intel/sof-tplg/sof-cht-rt5640.tplg
/usr/lib/firmware/intel/sof-tplg/sof-adl-rt711-l0-rt1316-l12-rt714-l3.tplg
/usr/lib/firmware/intel/sof-tplg/sof-ehl-rt5660.tplg
/usr/lib/firmware/intel/sof-tplg/sof-apl-pcm512x-master-44100.tplg
/usr/lib/firmware/intel/sof-tplg/sof-icl-es8336-dmic4ch-ssp2.tplg
/usr/lib/firmware/intel/sof-tplg/sof-apl-es8336-dmic2ch-ssp5.tplg
/usr/lib/firmware/intel/sof-tplg/sof-byt-rt5670-ssp0.tplg
/usr/lib/firmware/intel/sof-tplg/sof-hda-generic-idisp-4ch.tplg
/usr/lib/firmware/intel/sof-tplg/sof-mt8195-mt6359-rt1019-rt5682-dts.tplg
/usr/lib/firmware/intel/sof-tplg/sof-glk-rt5682.tplg
/usr/lib/firmware/intel/sof-tplg/sof-apl-rt298.tplg
/usr/lib/firmware/intel/sof-tplg/sof-glk-es8336-dmic4ch-ssp0.tplg
/usr/lib/firmware/intel/sof-tplg/sof-adl-rt711-l2-rt1316-l01-rt714-l3.tplg
/usr/lib/firmware/intel/sof-tplg/sof-jsl-nocodec.tplg
/usr/lib/firmware/intel/sof-tplg/sof-tgl-max98373-rt5682-xperi.tplg
/usr/lib/firmware/intel/sof-tplg/sof-adl-nocodec-ci.tplg
/usr/lib/firmware/intel/sof-tplg/sof-cht-nocodec.tplg
/usr/lib/firmware/intel/sof-tplg/sof-ehl-rt5660-nohdmi.tplg
/usr/lib/firmware/intel/sof-tplg/sof-apl-nocodec-ci.tplg
/usr/lib/firmware/intel/sof-tplg/sof-apl-es8336-dmic4ch-ssp0.tplg
/usr/lib/firmware/intel/sof-tplg/sof-byt-rt5640.tplg
/usr/lib/firmware/intel/sof-tplg/sof-glk-cs42l42.tplg
/usr/lib/firmware/intel/sof-tplg/sof-jsl-es8336-dmic4ch-ssp0.tplg
/usr/lib/firmware/intel/sof-tplg/sof-apl-pcm512x-master.tplg
/usr/lib/firmware/intel/sof-tplg/sof-icl-dmic-4ch.tplg
/usr/lib/firmware/intel/sof-tplg/sof-hda-generic-2ch-kwd.tplg
[... 247 lines cut ...]
/usr/lib/firmware/intel/sof/community/sof-cfl.ri
/usr/lib/firmware/intel/sof/community/sof-tgl-h.ri
/usr/lib/firmware/intel/sof/community/sof-cml.ri
/usr/lib/firmware/intel/sof/sof-byt.ri
/usr/lib/firmware/intel/sof/sof-rpl.ldc
/usr/lib/firmware/intel/sof/sof-icl.ri
/usr/lib/firmware/intel/sof/sof-jsl.ri
/usr/lib/firmware/intel/sof/sof-cnl.ri
/usr/lib/firmware/intel/sof/sof-adl.ldc
/usr/lib/firmware/intel/sof/sof-adl-n.ri
/usr/lib/firmware/intel/sof/sof-cml.ldc
/usr/lib/firmware/intel/sof/sof-rpl-s.ri
/usr/lib/firmware/intel/sof/sof-adl-s.ldc
/usr/lib/firmware/intel/sof/sof-icl.ldc
/usr/lib/firmware/intel/sof/sof-cfl.ldc
/usr/lib/firmware/intel/sof/sof-tgl.ri
/usr/lib/firmware/intel/sof/sof-ehl.ldc
/usr/lib/firmware/intel/sof/sof-tgl-h.ldc
/usr/lib/firmware/intel/sof/sof-glk.ri
/usr/lib/firmware/intel/sof/sof-cfl.ri
/usr/lib/firmware/intel/sof/sof-cht.ri
/usr/lib/firmware/intel/sof/sof-bdw.ri
/usr/lib/firmware/intel/sof/sof-jsl.ldc
/usr/lib/firmware/intel/sof/sof-tgl-h.ri
/usr/lib/firmware/intel/sof/sof-cml.ri

marc-hb, 2024-07-15:
> find / -xdev ( -nouser -o -nogroup )
> [...]
> have no owner or group.

That's not what `-nouser` means, check `man find`.

This is a problem with Debian packaging, please report this issue to Debian. @mrhpearson are you still packaging SOF for Debian?

The user and permissions in sof-bin.tgz file are random and should be discarded when packaging.

closed 2024-07-15 by marc-hb

mrhpearson, 2024-07-16:
Thanks for flagging this - yes I am still the packager.

I'm working on an update to the latest release but am a bit behind - I had it ready but found issues with the Meteorlake platforms not working and hadn't quite figured what had changed to break things...

I'll look at fixing this with the next update.

@MuldersMakes - I guess the correct way of doing this is a Debian bug for the issue? Or you can just wait for me to fix it ;) Up to you.

Mark

marc-hb, 2024-07-19:
> I had it ready but found issues with the Meteorlake platforms not working and hadn't quite figured what had changed to break things...

Maybe https://github.com/thesofproject/sof/issues/9243? I don't know whether that bug was ever in a release...

mrhpearson, 2024-07-19:
> > I had it ready but found issues with the Meteorlake platforms not working and hadn't quite figured what had changed to break things...
> 
> Maybe [thesofproject/sof#9243](https://github.com/thesofproject/sof/issues/9243)? I don't know whether that bug was ever in a release...

Nah - it was a packaging problem on my side. I need to go and refresh my memory but I think the MTL files moved and whatever I'm doing in the packaging somehow missed it or didn't account for it. This summer has been too hectic.

I see there's a new release out so I am going to target getting that done.

Mark