thesofproject/sof-bin #177: Missing topology file for audio on Zenbook S 14 with kernel 6.11

BlackJohnny, 2024-10-02:
Hi,

It may be useful to know that the topology file "intel/sof-ipc4-tplg/sof-lnl-cs42l43-l0-2ch.tplg" is requested during boot but is not present in the latest release. Is this file supposed to be included, or is the issue with another system part?

Thank you!

```
Oct 02 18:57:17 prime kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: Firmware paths/files for ipc type 1:
Oct 02 18:57:17 prime kernel: sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware file:     intel/sof-ipc4/lnl/sof-lnl.ri
Oct 02 18:57:17 prime kernel: sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware lib path: intel/sof-ipc4-lib/lnl
Oct 02 18:57:17 prime kernel: sof-audio-pci-intel-lnl 0000:00:1f.3:  Topology file:     intel/sof-ipc4-tplg/sof-lnl-cs42l43-l0-2ch.tplg
```

lgirdwood, 2024-10-15:
@ujfalusi @charleskeepax are we missing a Cirrus topology file ?

bardliao, 2024-10-16:
@lgirdwood sof-lnl-cs42l43-l0-2ch.tplg is not in SOF yet.

lgirdwood, 2024-10-16:
> @lgirdwood sof-lnl-cs42l43-l0-2ch.tplg is not in SOF yet.

@kv2019i @bardliao @charleskeepax lets make sure we get it merged for v2.12 then. Thanks!

BlackJohnny, 2024-10-16:
Update: the kernel 6.12-rc3 will not ask for "sof-lnl-cs42l43-l0-2ch.tplg" file instead will load "sof-lnl-cs42l43-l0-cs35l56-l23.tplg". The audio then works, not well but at least the board is detected, and we have sound.

lgirdwood, 2024-10-16:
> Update: the kernel 6.12-rc3 will not ask for "sof-lnl-cs42l43-l0-2ch.tplg" file instead will load "sof-lnl-cs42l43-l0-cs35l56-l23.tplg". The audio then works, not well but at least the board is detected, and we have sound.
 
@BlackJohnny ok, good to know - its probably expected not everything will work with a different topology file. Can you describe whats not working ?

BlackJohnny, 2024-10-17:
Functional:
- sound activation
- volume adjustment
- balance management

Malfunctions:
- sound stability is not good, occasionally transforming into a robotic tone for extended periods
- microphone input is not functional
- sound volume is insufficiently amplified

MatrixC7, 2025-03-27:
It seems that from Kernel 6.13.8 the problem is back. I have both error output with kernel version 6.13.8 and 6.14.0.
```
$ sudo dmesg | grep -i audio
[    3.722015] sof-audio-pci-intel-lnl 0000:00:1f.3: enabling device (0000 -> 0002)
[    3.722122] sof-audio-pci-intel-lnl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    4.360280] sof-audio-pci-intel-lnl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [xe])
[    4.366969] sof-audio-pci-intel-lnl 0000:00:1f.3: use msi interrupt mode
[    4.386047] sof-audio-pci-intel-lnl 0000:00:1f.3: hda codecs found, mask 4
[    4.386054] sof-audio-pci-intel-lnl 0000:00:1f.3: NHLT device BT(0) detected, ssp_mask 0x4
[    4.386055] sof-audio-pci-intel-lnl 0000:00:1f.3: BT link detected in NHLT tables: 0x4
[    4.386057] sof-audio-pci-intel-lnl 0000:00:1f.3: DMICs detected in NHLT tables: 2
[    4.388261] sof-audio-pci-intel-lnl 0000:00:1f.3: SOF firmware and/or topology file not found.
[    4.388265] sof-audio-pci-intel-lnl 0000:00:1f.3: Supported default profiles
[    4.388265] sof-audio-pci-intel-lnl 0000:00:1f.3: - ipc type 1 (Requested):
[    4.388266] sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware file: intel/sof-ipc4/lnl/sof-lnl.ri
[    4.388267] sof-audio-pci-intel-lnl 0000:00:1f.3:  Topology file: intel/sof-ipc4-tplg/sof-lnl-cs42l43-l0-cs35l56-l23-2ch.tplg
[    4.388268] sof-audio-pci-intel-lnl 0000:00:1f.3: Check if you have 'sof-firmware' package installed.
[    4.388268] sof-audio-pci-intel-lnl 0000:00:1f.3: Optionally it can be manually downloaded from:
[    4.388268] sof-audio-pci-intel-lnl 0000:00:1f.3:    https://github.com/thesofproject/sof-bin/
[    6.540124] sof-audio-pci-intel-lnl 0000:00:1f.3: error: sof_probe_work failed err: -2
```

ujfalusi, 2025-03-27:
@MatrixC7 , the new kernels will request the correct topology file, which is not yet released in sof-bin.

You can try to build the topologies from https://github.com/thesofproject/sof/ (./scripts/build-tools.sh -T) or you can try to install (after unzipping) [sof-lnl-cs42l43-l0-cs35l56-l23-2ch.tplg.zip](https://github.com/user-attachments/files/19480971/sof-lnl-cs42l43-l0-cs35l56-l23-2ch.tplg.zip) to /lib/firmware/intel/sof-ipc4-tplg/

kv2019i, 2025-03-27:
@MatrixC7  @ujfalusi I'm just working on a sof-bin-2025.01.1 release that will have sof-lnl-cs42l43-l0-cs35l56-l23-2ch.tplg .https://github.com/thesofproject/sof-bin/pull/189

PisonJay, 2025-03-27:
```log
❯ sudo dmesg | grep audio
[   11.643538] sof-audio-pci-intel-lnl 0000:00:1f.3: enabling device (0000 -> 0002)
[   11.643893] sof-audio-pci-intel-lnl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[   11.643979] sof-audio-pci-intel-lnl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [xe])
[   11.650495] sof-audio-pci-intel-lnl 0000:00:1f.3: use msi interrupt mode
[   11.669211] sof-audio-pci-intel-lnl 0000:00:1f.3: hda codecs found, mask 4
[   11.669219] sof-audio-pci-intel-lnl 0000:00:1f.3: NHLT device BT(0) detected, ssp_mask 0x4
[   11.669222] sof-audio-pci-intel-lnl 0000:00:1f.3: BT link detected in NHLT tables: 0x4
[   11.669226] sof-audio-pci-intel-lnl 0000:00:1f.3: DMICs detected in NHLT tables: 4
[   11.674425] sof-audio-pci-intel-lnl 0000:00:1f.3: SOF firmware and/or topology file not found.
[   11.674436] sof-audio-pci-intel-lnl 0000:00:1f.3: Supported default profiles
[   11.674438] sof-audio-pci-intel-lnl 0000:00:1f.3: - ipc type 1 (Requested):
[   11.674442] sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware file: intel/sof-ipc4/lnl/sof-lnl.ri
[   11.674444] sof-audio-pci-intel-lnl 0000:00:1f.3:  Topology file: intel/sof-ipc4-tplg/sof-lnl-cs42l43-l0-cs35l56-l23-4ch.tplg
[   11.674446] sof-audio-pci-intel-lnl 0000:00:1f.3: Check if you have 'sof-firmware' package installed.
[   11.674448] sof-audio-pci-intel-lnl 0000:00:1f.3: Optionally it can be manually downloaded from:
[   11.674449] sof-audio-pci-intel-lnl 0000:00:1f.3:    https://github.com/thesofproject/sof-bin/
[   18.227549] sof-audio-pci-intel-lnl 0000:00:1f.3: error: sof_probe_work failed err: -2
```

For my case the kernel requests for `intel/sof-ipc4-tplg/sof-lnl-cs42l43-l0-cs35l56-l23-4ch.tplg` on Zenbook S 16. Can I build the topology from https://github.com/thesofproject/sof/ just like `sof-lnl-cs42l43-l0-cs35l56-l23-2ch.tplg`? I'm not sure about it.

ujfalusi, 2025-03-27:
@PisonJay, is this works for you:  [sof-lnl-cs42l43-l0-cs35l56-l23-4ch.tplg.zip](https://github.com/user-attachments/files/19483963/sof-lnl-cs42l43-l0-cs35l56-l23-4ch.tplg.zip)

singalsu, 2025-03-27:
Please try to build topology with this patch:

```
diff --git a/tools/topology/topology2/production/tplg-targets-ace2.cmake b/tools/topology/topology2/production/tplg-targets-ace2.cmake
index 12a75b3bb..ba23fb4b5 100644
--- a/tools/topology/topology2/production/tplg-targets-ace2.cmake
+++ b/tools/topology/topology2/production/tplg-targets-ace2.cmake
@@ -80,6 +80,14 @@ SDW_JACK_OUT_STREAM=Playback-SimpleJack,SDW_JACK_IN_STREAM=Capture-SimpleJack,\
 HDMI1_ID=5,HDMI2_ID=6,HDMI3_ID=7,DMIC0_ENHANCED_CAPTURE=true,\
 EFX_DMIC0_TDFB_PARAMS=line2_generic_pm10deg,EFX_DMIC0_DRC_PARAMS=dmic_default"
 
+"cavs-sdw\;sof-lnl-cs42l43-l0-cs35l56-l3-4ch\;PLATFORM=lnl,\
+NUM_DMICS=4,PDM1_MIC_A_ENABLE=1,PDM1_MIC_B_ENABLE=1,DMIC0_ID=3,DMIC1_ID=4,\
+PREPROCESS_PLUGINS=nhlt,NHLT_BIN=sof-lnl-cs42l43-l0-cs35l56-l3-4ch.bin,\
+NUM_SDW_AMP_LINKS=1,SDW_AMP_FEEDBACK=false,SDW_SPK_STREAM=Playback-SmartAmp,\
+SDW_JACK_OUT_STREAM=Playback-SimpleJack,SDW_JACK_IN_STREAM=Capture-SimpleJack,\
+HDMI1_ID=5,HDMI2_ID=6,HDMI3_ID=7,DMIC0_ENHANCED_CAPTURE=true,\
+EFX_DMIC0_TDFB_PARAMS=line4_pass,EFX_DMIC0_DRC_PARAMS=dmic_default"
+
 "cavs-sdw\;sof-lnl-cs42l43-l0-cs35l56-l23\;PLATFORM=lnl,NUM_SDW_AMP_LINKS=2,SDW_DMIC=1,\
 SDW_AMP_FEEDBACK=false,SDW_SPK_STREAM=Playback-SmartAmp,SDW_DMIC_STREAM=Capture-SmartMic,\
 SDW_JACK_OUT_STREAM=Playback-SimpleJack,SDW_JACK_IN_STREAM=Capture-SimpleJack"

```

Need to change the beamformer blob since we don't have a generic setup for four microphones arrays. Blob "line4_pass" disables the feature.

PisonJay, 2025-03-27:
@singalsu and @ujfalusi, thanks for quick response! I have tried both prebuilt tplgs posted above and self-built tplgs with the cmake patch (there's a little typo that `l3` should be `l23` instead). But with either tlpg the dmesg yields the same error that 'dmic01' not exist.

```
❯ sudo dmesg | grep -E '(audio|sof)'
[    0.043918] software IO TLB: area num 8.
[    0.716818] PCI-DMA: Using software bounce buffering for IO (SWIOTLB)
[    0.716819] software IO TLB: mapped [mem 0x0000000050000000-0x0000000054000000] (64MB)
[    9.502201] sof-audio-pci-intel-lnl 0000:00:1f.3: enabling device (0000 -> 0002)
[    9.502305] sof-audio-pci-intel-lnl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    9.502346] sof-audio-pci-intel-lnl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [xe])
[    9.509275] sof-audio-pci-intel-lnl 0000:00:1f.3: use msi interrupt mode
[    9.526383] sof-audio-pci-intel-lnl 0000:00:1f.3: hda codecs found, mask 4
[    9.526386] sof-audio-pci-intel-lnl 0000:00:1f.3: NHLT device BT(0) detected, ssp_mask 0x4
[    9.526387] sof-audio-pci-intel-lnl 0000:00:1f.3: BT link detected in NHLT tables: 0x4
[    9.526388] sof-audio-pci-intel-lnl 0000:00:1f.3: DMICs detected in NHLT tables: 4
[    9.528971] sof-audio-pci-intel-lnl 0000:00:1f.3: Firmware paths/files for ipc type 1:
[    9.528974] sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware file:     intel/sof-ipc4/lnl/sof-lnl.ri
[    9.528975] sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware lib path: intel/sof-ipc4-lib/lnl
[    9.528975] sof-audio-pci-intel-lnl 0000:00:1f.3:  Topology file:     intel/sof-ipc4-tplg/sof-lnl-cs42l43-l0-cs35l56-l23-4ch.tplg
[    9.529281] sof-audio-pci-intel-lnl 0000:00:1f.3: Loaded firmware library: ADSPFW, version: 2.12.0.1
[    9.776173] sof-audio-pci-intel-lnl 0000:00:1f.3: Booted firmware version: 2.12.0.1
[    9.784049] sof-audio-pci-intel-lnl 0000:00:1f.3: Topology: ABI 3:29:1 Kernel ABI 3:23:1
[    9.784258] sof_sdw sof_sdw: ASoC: physical link dmic01 (id 3) not exist
[    9.784259] sof_sdw sof_sdw: ASoC: topology: could not load header: -22
[    9.784295] sof-audio-pci-intel-lnl 0000:00:1f.3: error: tplg component load failed -22
[    9.784297] sof-audio-pci-intel-lnl 0000:00:1f.3: error: failed to load DSP topology -22
[    9.784299] sof-audio-pci-intel-lnl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[    9.784310] sof_sdw sof_sdw: ASoC: failed to instantiate card -22
[    9.785282] sof_sdw sof_sdw: error -EINVAL: snd_soc_register_card failed -22
[    9.785286] sof_sdw sof_sdw: probe with driver sof_sdw failed with error -22
```

All the results above are with kernel 6.13.8.

When with kernel 6.13.7, I could have playback working, but I didn't test mics.

PisonJay, 2025-03-27:
update:

with kernel 6.13.7, it loads `intel/sof-ipc4-tplg/sof-lnl-cs42l43-l0-cs35l56-l23.tplg`. Playback works but capture is not working.

```
❯ sudo dmesg | grep -E '(audio|sof)'
[    0.043230] software IO TLB: area num 8.
[    0.716632] PCI-DMA: Using software bounce buffering for IO (SWIOTLB)
[    0.716632] software IO TLB: mapped [mem 0x0000000050040000-0x0000000054040000] (64MB)
[    9.765443] sof-audio-pci-intel-lnl 0000:00:1f.3: enabling device (0000 -> 0002)
[    9.765550] sof-audio-pci-intel-lnl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    9.765594] sof-audio-pci-intel-lnl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [xe])
[    9.772428] sof-audio-pci-intel-lnl 0000:00:1f.3: use msi interrupt mode
[    9.791153] sof-audio-pci-intel-lnl 0000:00:1f.3: hda codecs found, mask 4
[    9.791157] sof-audio-pci-intel-lnl 0000:00:1f.3: NHLT device BT(0) detected, ssp_mask 0x4
[    9.791158] sof-audio-pci-intel-lnl 0000:00:1f.3: BT link detected in NHLT tables: 0x4
[    9.791159] sof-audio-pci-intel-lnl 0000:00:1f.3: DMICs detected in NHLT tables: 4
[    9.793763] sof-audio-pci-intel-lnl 0000:00:1f.3: Firmware paths/files for ipc type 1:
[    9.793765] sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware file:     intel/sof-ipc4/lnl/sof-lnl.ri
[    9.793766] sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware lib path: intel/sof-ipc4-lib/lnl
[    9.793767] sof-audio-pci-intel-lnl 0000:00:1f.3:  Topology file:     intel/sof-ipc4-tplg/sof-lnl-cs42l43-l0-cs35l56-l23.tplg
[    9.794063] sof-audio-pci-intel-lnl 0000:00:1f.3: Loaded firmware library: ADSPFW, version: 2.12.0.1
[   10.051427] sof-audio-pci-intel-lnl 0000:00:1f.3: Booted firmware version: 2.12.0.1
[   10.063821] sof-audio-pci-intel-lnl 0000:00:1f.3: Topology: ABI 3:29:1 Kernel ABI 3:23:1
[   10.064144] sof_sdw sof_sdw: ASoC: Parent card not yet available, widget card binding deferred
[   10.064809] sof-audio-pci-intel-lnl 0000:00:1f.3: Loaded firmware library: ADSPFW, version: 2.12.0.1
[   10.099690] sof_sdw sof_sdw: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 3
[   11.211857] input: sof-soundwire Jack as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input16
[   11.211897] input: sof-soundwire HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input17
[   11.211922] input: sof-soundwire HDMI/DP,pcm=6 as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input18
[   11.211946] input: sof-soundwire HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input19
```

ujfalusi, 2025-03-27:
@PisonJay , can you  try this: [sof-lnl-cs42l43-l0-cs35l56-l23-4ch.tplg (2).zip](https://github.com/user-attachments/files/19484812/sof-lnl-cs42l43-l0-cs35l56-l23-4ch.tplg.2.zip) unzip it and copy to /lib/firmware/intel/sof-ipc4-tplg/

PisonJay, 2025-03-27:
> [@PisonJay](https://github.com/PisonJay) , can you try this: [sof-lnl-cs42l43-l0-cs35l56-l23-4ch.tplg (2).zip](https://github.com/user-attachments/files/19484812/sof-lnl-cs42l43-l0-cs35l56-l23-4ch.tplg.2.zip) unzip it and copy to /lib/firmware/intel/sof-ipc4-tplg/

the result is the same, physical link dmic01 (id 3) not exist

```
❯ sudo dmesg | grep -E '(audio|sof)'
[    0.043901] software IO TLB: area num 8.
[    0.725487] PCI-DMA: Using software bounce buffering for IO (SWIOTLB)
[    0.725488] software IO TLB: mapped [mem 0x0000000050001000-0x0000000054001000] (64MB)
[   10.054389] sof-audio-pci-intel-lnl 0000:00:1f.3: enabling device (0000 -> 0002)
[   10.054524] sof-audio-pci-intel-lnl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[   10.054565] sof-audio-pci-intel-lnl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [xe])
[   10.061475] sof-audio-pci-intel-lnl 0000:00:1f.3: use msi interrupt mode
[   10.079641] sof-audio-pci-intel-lnl 0000:00:1f.3: hda codecs found, mask 4
[   10.079645] sof-audio-pci-intel-lnl 0000:00:1f.3: NHLT device BT(0) detected, ssp_mask 0x4
[   10.079646] sof-audio-pci-intel-lnl 0000:00:1f.3: BT link detected in NHLT tables: 0x4
[   10.079648] sof-audio-pci-intel-lnl 0000:00:1f.3: DMICs detected in NHLT tables: 4
[   10.082147] sof-audio-pci-intel-lnl 0000:00:1f.3: Firmware paths/files for ipc type 1:
[   10.082149] sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware file:     intel/sof-ipc4/lnl/sof-lnl.ri
[   10.082149] sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware lib path: intel/sof-ipc4-lib/lnl
[   10.082150] sof-audio-pci-intel-lnl 0000:00:1f.3:  Topology file:     intel/sof-ipc4-tplg/sof-lnl-cs42l43-l0-cs35l56-l23-4ch.tplg
[   10.082446] sof-audio-pci-intel-lnl 0000:00:1f.3: Loaded firmware library: ADSPFW, version: 2.12.0.1
[   10.344345] sof-audio-pci-intel-lnl 0000:00:1f.3: Booted firmware version: 2.12.0.1
[   10.354986] snd_soc_sdw_utils:asoc_sdw_count_sdw_endpoints: sof_sdw sof_sdw: Found 5 devices with 8 endpoints
[   10.354991] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Adding prefix cs42l43 for cs42l43-codec
[   10.354993] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Add dev: 0, 0x3001fa424301 end: 0, dai: 0, P/- to solo: 0
[   10.354995] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Add dev: 0, 0x3001fa424301 end: 1, dai: 2, -/C to solo: 0
[   10.354997] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Add dev: 0, 0x3001fa424301 end: 2, dai: 0, -/C to solo: 0
[   10.354999] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Adding prefix AMP1 for sdw:0:2:01fa:3556:01:0
[   10.355000] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Add dev: 2, 0x23001fa355601 end: 0, dai: 1, P/C to group: 1
[   10.355002] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Adding prefix AMP2 for sdw:0:2:01fa:3556:01:1
[   10.355003] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Add dev: 2, 0x23101fa355601 end: 0, dai: 1, P/C to group: 1
[   10.355004] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Adding prefix AMP3 for sdw:0:3:01fa:3556:01:2
[   10.355005] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Add dev: 3, 0x33201fa355601 end: 0, dai: 1, P/C to group: 1
[   10.355006] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Adding prefix AMP4 for sdw:0:3:01fa:3556:01:3
[   10.355007] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Add dev: 3, 0x33301fa355601 end: 0, dai: 1, P/C to group: 1
[   10.355009] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW0-Playback-SimpleJack, id 0
[   10.355011] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW0-Capture-SmartMic, id 4
[   10.355013] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW0-Capture-SimpleJack, id 1
[   10.355014] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW2-Playback-SmartAmp, id 2
[   10.355016] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW2-Capture-SmartAmp, id 3
[   10.355017] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link dmic01, id 5
[   10.355018] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link dmic16k, id 6
[   10.355019] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link iDisp1, id 7
[   10.355020] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link iDisp2, id 8
[   10.355021] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link iDisp3, id 9
[   10.356482] sof-audio-pci-intel-lnl 0000:00:1f.3: Topology: ABI 3:29:1 Kernel ABI 3:23:1
[   10.356771] sof_sdw sof_sdw: ASoC: physical link dmic01 (id 3) not exist
[   10.356773] sof_sdw sof_sdw: ASoC: topology: could not load header: -22
[   10.356818] sof-audio-pci-intel-lnl 0000:00:1f.3: error: tplg component load failed -22
[   10.356822] sof-audio-pci-intel-lnl 0000:00:1f.3: error: failed to load DSP topology -22
[   10.356823] sof-audio-pci-intel-lnl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[   10.356838] sof_sdw sof_sdw: ASoC: failed to instantiate card -22
[   10.357841] sof_sdw sof_sdw: error -EINVAL: snd_soc_register_card failed -22
[   10.357848] sof_sdw sof_sdw: probe with driver sof_sdw failed with error -22
```

ujfalusi, 2025-03-27:
@PisonJay, yes, please open a new issue fro S16 and let this one to be closed when @MatrixC7 have time to verify the topology file and when the sof-bin-2025.01.1 is released.

PisonJay, 2025-03-27:
@ujfalusi I opened a new issue at https://github.com/thesofproject/sof/issues/9930

MatrixC7, 2025-03-27:
@ujfalusi It seems that with topology file the firmware is loaded but in system there is still no available device for sound output. In LNL-M HD Audio Controller we do not have the 'Play HiFi quailty Music' option but only 'Pro Audio'.

[image]

ujfalusi, 2025-03-28:
@MatrixC7, I'm glad to hear that now the audio is up!
The issue is now in user space, likely if you update UCM (https://github.com/alsa-project/alsa-ucm-conf/) things will start to work in UI level as well.

It also worth a try to update pulseaudio/pipewire.

@charleskeepax, I believe all the pieces are in upstream for this in UCM?

libbrfish, 2025-04-10:
I'm having the same problem as @MatrixC7. Did you manage to fix this?
I have the latest alsa-ucm-conf but no sound and only a dummy output.

MatrixC7, 2025-04-10:
Mine is fixed with putting `ucm2` folder to its place of latest repository of alsa-ucm-conf. 'Play HiFi quailty Music' is back and the output of sound is normal.
The only problem is while using Proton or Wine to run .exe programs the sound output of LNL sound card would stop after running for several seconds. It exists both in the previous kernel or the current one.

ujfalusi, 2025-04-11:
@MatrixC7, can you try what is described in https://github.com/thesofproject/linux/issues/5284#issuecomment-2576977535 and see if that helps?